RaspAP/raspap-webgui #498: Is there a way to only have the web interface be shown?

Piflyer, 2020-02-19:
I'm trying to have a web panel to control stuff that is easily accessible by a touchscreen. I don't need the hostapd, so is there an option where the web ui is the only thing running?  I tried not downloading the hostapd, but the web ui will not show up.

billz, 2020-02-19:
Have you tried enabling [monitor mode](https://github.com/billz/raspap-webgui/wiki/FAQs#monitor)?

Piflyer, 2020-02-20:
Yeah, but I'm looking forward to being able to use the touchscreen to change WiFi settings and such, but without the hotspot. I tried avoiding the hostapd, and just running lighttpd with whatever was in the repo, but it gave me an error 500

billz, 2020-02-20:
Use the quick installer as usual, followed by `sudo apt-get remove hostapd`. The UI will continue to work and you can save settings, but the service does not exist.

Piflyer, 2020-02-20:
Thanks, that worked for me!

closed 2020-02-20 by Piflyer