RaspAP/raspap-webgui #508: My raspi download throw eth0 instead of tun0, is it normal?

Thatoo, 2020-02-29:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
All my computer connected by wifi to my raspi4 benefit from the OpenVPN throw tun0 said differently, it seems that my raspi4 forward well all traffic from OpenVPN's tun0 interface to my configured wireless interface (wlan0). However, if I connect through ssh to my raspi4 and I download something like wget https://xxxx.yyyy/file.data, it download throw eth0 instead of tun0.
How can I make sure that all traffic on my raspi4 are also going throw tun0?

**Your environment**
* Raspberry Pi 4
* Raspbian version : Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install 
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset
* Other software or services running with RaspAP? No

**Expected behavior**
How can I make sure that all traffic on my raspi4 are also going throw tun0?

**Actual behavior**
My raspi4 internet traffic go throw eth0 instead of tun0.

**Screenshots**
If applicable, add screenshots to help explain your problem.

billz, 2020-03-09:
RaspAP handles NAT'ing of traffic on tun0 for you, but not eth0. Masquerade outgoing traffic with:
```
iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
```

RaspAP forwards traffic for wlan0. You will need additional rules for eth0:

```
iptables -A FORWARD -i tun0 -o eth0 -m state --state RELATED,ESTABLISHED -j ACCEPT 
iptables -A FORWARD -i eth0 -o tun0 -j ACCEPT 
```

Finally, allow return traffic on these interfaces:
```
iptables -A INPUT -i eth0 -m state --state RELATED,ESTABLISHED -j ACCEPT
iptables -A INPUT -i tun0 -m state --state RELATED,ESTABLISHED -j ACCEPT
```
With the openvpn-client service running, check `netstat -nr` and `curl ifconfig.me`

```
$ netstat -nr
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         10.8.1.1        128.0.0.0       UG        0 0          0 tun0
0.0.0.0         192.168.1.254   0.0.0.0         UG        0 0          0 eth0
0.0.0.0         10.3.141.1      0.0.0.0         UG        0 0          0 wlan0
10.3.141.0      0.0.0.0         255.255.255.0   U         0 0          0 wlan0
10.8.1.0        0.0.0.0         255.255.255.0   U         0 0          0 tun0
128.0.0.0       10.8.1.1        128.0.0.0       UG        0 0          0 tun0
170.130.139.82  192.168.1.254   255.255.255.255 UGH       0 0          0 eth0
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0
```
```
$ curl ifconfig.me
170.130.139.83
```

billz, 2020-03-12:
Closing this as answered. Feel free to report your results

closed 2020-03-12 by billz