RaspAP/raspap-webgui #509: Error ‘cannot execute „wpa_cli reconfigure“‘

metawops, 2020-02-29:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
When trying to connect to a WiFi as client I get the error `cannot execute “wpa_cli reconfigure“`.

**Your environment**
* Raspberry Pi 4B 1.1 4GB
* Raspbian version: Buster
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Yes (without openVPN)
* Simultaneous AP and managed mode? Yes
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**
- Click on „Configure WiFi client“ in the menu.
- Click on the „Connect“ button of an existing and found WiFi.

**Expected behavior**
The connection is made.

**Actual behavior**
No connection is made, instead the above mentioned error occurs.

**Screenshots**
[image]

**Additional context**
When issued the command `wpa_cli reconfigure` on the command line I get the error shown in the second screenshot: `Failed to connect to non-global ctrl_ifname: (nil). error: No such file or directory`

[image]

The `/etc/sudoers` file contains these `wpa_cli` related lines:
[image]

billz, 2020-02-29:
https://github.com/billz/raspap-webgui/wiki/FAQs#scanning

metawops, 2020-02-29:
Cool! That made the error go away! 🙂 
Sorry I didn‘t find that in the FAQ – the reason being that the headline/question doesn‘t apply to my problem: WiFi scanning worked fine for me.

billz, 2020-02-29:
Glad to hear it. I'll change the title in the FAQ to include this wpa_supplicant error, thanks.

closed 2020-02-29 by billz

slayerkodi, 2023-05-08:
Is the same for a pi400?