RaspAP/raspap-webgui #512: sudo installers/uninstall.sh does not work

bkrajendra, 2020-03-01:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I am trying to remove RaspAP by using:
sudo installers/uninstall.sh
but im getting error:
` ./uninstall.sh: 30: syntax error : "(" unexpected '

**Your environment**
* Raspberry Pi 3 Model B+ 
* Raspbian version:  Buster Desktop
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install 
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (N)
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**
sudo sh ./uninstall.sh

billz, 2020-03-01:
Use ` sudo bash ./uninstall.sh` or simply `sudo ./uninstall.sh`

bkrajendra, 2020-03-01:
Thanks @billz 
Anyways it worked with same command ..
I just edited with nano uninstall.sh saved it and it worked.
I don't know how but finally it got it removed and all my settings are back.

This is my 4th time i was giving a try to `raspap-webgui`
It's a really nice functionality... but everytime I try, i get disappointed !

This time everything worked fine, but as soon as I switched on AP mode everything got messed.
AP was not allowing to connect, server was not working. wifi client got disconnected from main wifi.
I connected my device to Monitor and I tried a lot from there to connect to local wifi, but it did not worked.
Constantly i was getting error wpa_cli reconfigure not working!

Finally I had to remove everything.
I will definitely give it a try later sometime.

I think to make it more stable and standard can we follow some other implementations.
Which are very simple but stable and never creates a trouble.
For example i have seen a RPI based implementation called "[Volumio](https://volumio.org/get-started/)" 
They uses wifi configuration through AP, but it's very slick and simple and works!

This is just a suggestion.. 
otherwise it's the best tool available.
Many people are trying it and working fine which means i am doing something wrong..!

billz, 2020-03-01:
wpa_supplicant and hostapd are sometimes problematic. this may help https://github.com/billz/raspap-webgui/wiki/FAQs#scanning

closed 2020-03-01 by billz