RaspAP/raspap-webgui #513: Custom hostname doesn't work with cert quick installer

bmitchinson, 2020-03-02:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
After changing the hostname, following the quick install then cert quick install, http://{hostname}.local works but not https://{hostname}.local

**Your environment**
* Raspberry Pi hardware: (Pi 3 Model B+)  
* Raspbian version: (Buster Lite)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? (Quick)
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Wifi client AP Mode)
* Onboard wireless chipset or external adapter? (Onboard)
* Other software or services running with RaspAP? (None)

**Steps to reproduce**
- flashing a fresh Raspbian Buster Lite
- apt-get update, apt-get upgrade
- raspi-config to set the country code
- raspi-config to set the hostname
- `curl -sL https://install.raspap.com | bash` y to all.
- changed ssid and password
- `curl -sL https://install.raspap.com | bash -s -- --cert` y to all.
- on mac osx, installed the cert by dragging it to System and choosing Always Trust, then rebooted.
- when visiting http://{hostname.local} I get the config as expected
- when visiting https://{hostname.local}:443 I get the "your connection is not private"

**Expected behavior**
Visiting {hostname.local}:443 connects to the default config site

**Actual behavior**
Visiting {hostname.local}:443 gives  "your connection is not private".

bmitchinson, 2020-03-02:
Does the cert fuffil the newest catalina requirements? didn't work on iOS 13 for me either?

https://superuser.com/questions/1492207/neterr-cert-revoked-in-chrome-chromium-introduced-with-macos-catalina

bmitchinson, 2020-03-02:
I was able to follow this whack guide: https://css-tricks.com/getting-around-revoked-certificate-osx/

Which gave me a "pi@{hostname}.cer" cert? Should I serve that off the pi maybe.

bmitchinson, 2020-03-02:
wasn't able to install that .cer on an iphone and then use it to access, but did work on the mac that made it from the "browser drag to desktop and install" trick.

billz, 2020-03-02:
Interesting, looks like this is indeed an [issue with Catalina](https://github.com/FiloSottile/mkcert/issues/174). I ran through your steps on Mojave and got a valid cert.

[image]

bmitchinson, 2020-03-02:
Appreciate you looking at this so quickly! Would be curious if that is also the cause for an iPhone on iOS13? looks like it is?

billz, 2020-03-02:
This is apparently [fixed in v.1.4.0](https://github.com/FiloSottile/mkcert/issues/174#issuecomment-522157570) of mkcert. This project uses v.1.3. I'll update the installer shortly to see if that resolves it. Thanks for reporting

bmitchinson, 2020-03-02:
Testing now

bmitchinson, 2020-03-02:
Worked perfectly! Closing.

billz, 2020-03-02:
Great, glad to hear it

bmitchinson, 2020-03-02:
Quick Installer script worked perfectly! Thanks for such a speedy fix! Looking forward to using this here: https://github.com/Haiti-Distributed-Healthcare-System/hdhs

We're early in development, but I'll be sure to discuss sponsorship once we're a bit further along! Appreciate it.

billz, 2020-03-02:
@bmitchinson looks like an outstanding initiative! Honestly, I'd be happy just knowing some part of this project proved useful for a non-profit/humanitarian mission. Maybe down the road you'd consider sharing it on https://github.com/billz/raspap-awesome

bmitchinson, 2020-03-03:
@billz , anyway you or anyone else could test this with an iOS device 13.3.1 (Most recent)? Still isn't working for my iPhone even though catalina is fine.

https://github.com/FiloSottile/mkcert/issues/202

billz, 2020-03-03:
@bmitchinson followed these steps and obtained a valid cert on iOS 1.3.1 https://support.apple.com/en-nz/HT204477

[image]
[image]
[image]

bmitchinson, 2020-03-03:
Didn't know about that additional "trust cert" screen. Man that's annoyingly long to go through...
Thank you!

closed 2020-03-03 by bmitchinson

billz, 2020-03-03:
It _is_ annoying... but I suppose also mirrors the rootCA trust steps on macOS