RaspAP/raspap-webgui #514: Wireguard client support

billz, 2020-03-02:
Wireguard was recently merged into the Linux kernel. RaspAP can/should offer a Wireguard install + client config as an alternative to OpenVPN, even before it makes it into a future Raspbian distro.

firestrife23, 2020-04-04:
I was gonna make a request but I'm glad to see it's on the list for future enhancement. Following this for updates.

imc67, 2020-04-14:
It would be awesome to have this implemented in RaspAP! I just moved over to WireGuard (PiVPN) and it’s really easy, fast and the iOS and macOS apps are really really good (always connected, auto connect after reboot or flightmodus, etc).

billz, 2020-04-22:
Installer portion is mostly done. Frontend in progress. 

[image]

Note the custom WireGuard icon font ;)
Development may be tracked on the `feature/wireguard` branch.

imc67, 2020-04-22:
@billz this is awesome! I’m not sure how exactly you implemented it, is RaspAP a WireGuard server for clients in its local network or is RaspAP a client with a connection to a WireGuard server anywhere on the internet to create a save tunnel to your home? The latter is what I would really appreciate to have.

Keep up your great work!

CX32, 2020-07-06:
@billz hi sorry to bother, but is there any progress on having this in master?

pietrondo, 2020-10-22:
news about wireguard support?

jimmont, 2021-01-13:
@billz if I can help with the frontend (primarily) let me know what to branch from, remaining work items and how best to communicate;

rusagent, 2021-01-14:
I can maybe help as well!

billz, 2021-01-14:
@jimmont @rusagent much appreciated. There's quite a lot pending at the moment, including the 2.6-beta. Once this is released I can again focus on the `feature/wireguard` branch. 

In the meantime, you're welcome to join the beta testing and/or pull the feature branch to help push things forward. Thanks.

rusagent, 2021-01-14:
@billz  I got the beta installed right now and I'm having quite some trouble after. I'll investigate further into it what causes my problems. 
And do you have a roadmap or any guidelines on how to proceed with the wireguard branch? I'd really finally do something useful with my coding skills^^

billz, 2021-03-07:
WireGuard support is (finally!) being released as [an experimental feature](https://docs.raspap.com/wireguard/).
The video walkthrough below shows just how easily a WireGuard VPN tunnel can be created with RaspAP. 

https://user-images.githubusercontent.com/229399/110251359-63182400-7f80-11eb-98d0-2055a38fccbc.mov

I've invested a lot of time and effort to make this as streamlined and user-friendly as possible. Attention to detail is here too, such as creating a custom font with the WireGuard logo, which isn't included in Font Awesome's library.

For these reasons, WireGuard will be released initially as an exclusive feature for [Insiders](https://docs.raspap.com/insiders/). After the first funding target is met, WireGuard and a bundle of other awesome stuff will be merged into the public RaspAP repository. This is all done with full transparency through our fiscal host, [OpenCollective](https://opencollective.com/raspap).

If you'd like to have early access to WireGuard (or you just love RaspAP) I'd encourage you to support our work by [joining Insiders](https://github.com/sponsors/RaspAP/).

More info on WireGuard for RaspAP: https://docs.raspap.com/wireguard/

imc67, 2021-03-08:
Wow, @billz you've done a really great job and created one of the first and most practical GUI's for WG.

In this comment on 22-4-2020: https://github.com/RaspAP/raspap-webgui/issues/514?_pjax=%23js-repo-pjax-container#issuecomment-617837265 I was wondering if you would create a WG server or WG client. Am I correct that you only created the WG server? 

I want to use RaspAP as a travel router and want to be always connected to a WG server, is that possible now?

billz, 2021-03-08:
@imc67 good question. This demonstrates creating both a server config (local to the Pi) and a client config for a remote peer. 
In your case, given that you have an existing WG server, you would configure your Pi as the peer, choose the "Enable endpoint" option and start WG. 

WG is somewhat agnostic when it comes to the server vs. peer distinction. That is, the only difference is in the configuration. This is roughly analogous to how SSH key exchange works. Hope this makes sense.

closed 2021-11-27 by billz