RaspAP/raspap-webgui #515: development process

glaszig, 2020-03-02:
i'm sort of proficient with vim but not very efficient. so i have my working copy of raspap on my workstation and automatically copy changes over to the pi. i've written the following script to automate this process by way of `fswatch` and just wanted to share. but i'm unsure if and how we want to have this part of documentation or wiki.

https://gist.github.com/glaszig/673691a6ba7bdc8e3d054d826a179976

billz, 2020-03-03:
This looks great. I've always used vim + tmux on the Pi, mostly out of habit. But I can see this being a viable alternative. Would it make sense to include a link to a new wiki page under [how to contribute](https://github.com/billz/raspap-webgui#how-to-contribute)?

glaszig, 2020-03-03:
go ahead if you‘d like. in any case, this can be closed then.

billz, 2020-03-25:
Added here https://github.com/billz/raspap-webgui/wiki/Development-process
Will update the readme with a link. thanks again

closed 2020-03-25 by billz