RaspAP/raspap-webgui #516: Need a walk thought.

Whitemanns, 2020-03-03:
Hello The walk though of this is not detailed to where i understand it. It tell me how to install it but not set it up. I can get the Raspapa to accept my internet but not let me then "hotspot" it and connect to it.

bmitchinson, 2020-03-03:
Hi @Whitemanns , did you use the quick installer described here: https://raspap.com/#quick-installer ?

After doing this once, if you reboot your pi you can connect with the information in that guide, also pictured here:

[image]

Whitemanns, 2020-03-03:
So the problem is I can get the pie setup with Raspap I can connect to the console of it and I can get it to accept my Wi-Fi but when I try to use the hotspot to put out a signal it does but I cannot connect to it it says cannot obtain IP I have reinstall the OS and Raspap 7 times now when I look through the log it says hostapd error code 1.

billz, 2020-03-05:
Please be sure to follow the project [prerequisites](https://github.com/billz/raspap-webgui#prerequisites). RaspAP's quick installer has been validated to work on all currently available Pi's with onboard wireless. Start with the default config before changing settings. Unabling logging with hostapd is useful for troubleshooting. Refer to the [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs) and [reporting issues](https://github.com/billz/raspap-webgui/wiki/Reporting-issues) if not done so. Thanks

closed 2020-03-05 by billz