RaspAP/raspap-webgui #524: Hostpad and Wifi globally not working in Raspian Fresh install

thecalle, 2020-03-09:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

Raspbian SD fresh, nothing installed, Raspi-Config never launched. Only touch /boot/ssh file to enable SSH Access. 

**My environment**
* Raspberry Pi hardware: Pi 3 Model B+, no USB adapters, LAN cable connected
* Raspbian version: 2020-02-13-raspbian-buster-lite
* Followed the project prerequisites
* Checked the project FAQ.
* RaspAP Quick Install
* Using default configuration, nothing touched at ALL

**Steps to reproduce**
1) Burn SD
2) touch SSH file in /boot/
3) Start Raspberry with LAN connected
4) Connect using SSH and start
5) Run autoinstall script: curl -sL https://install.raspap.com | bash -s -- --yes
6) Reboot

ISSUES:
1) Hostapd fails to start
2) Wpa_supplicant started but shell access notifies: Wi-Fi is currently blocked by rfkill

Can't connect to hostapd to configure device.

billz, 2020-03-09:
> Raspi-Config never launched

Setting your wifi country with raspi-config is part of the [project prerequisites](https://github.com/billz/raspap-webgui#prerequisites), along with a updating your Pi's kernel and firmware. Did you follow these steps before running the installer?

thecalle, 2020-03-09:
I can confirm, using pre-requisites, hostapd starts. Wifi scan not working.
Wpa_supplicant service must be modified with:

ExecStart=/sbin/wpa_supplicant -u -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0

closed 2020-03-09 by thecalle

billz, 2020-03-09:
Thanks for including this. wpa_supplicant can be problematic with hostapd, so we may end up including something like this with the installer.