RaspAP/raspap-webgui #533: raspap with proxy

marafa, 2020-03-18:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.
- i want to use a proxy with raspap - how do i configure raspap to use the proxy?

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  pi 3
* Raspbian version (examples: Buster Lite, Buster Desktop) buster  lite
* Followed the project prerequisites? (Y/N) y
* Checked the project FAQ? (Y/N) y
* RaspAP Quick Install or Manual setup? quick
* Using default configuration? (Y/N) y
* Simultaneous AP and managed mode? (Y/N) what ever is the default
* Onboard wireless chipset or external adapter? onboard
* Other software or services running with RaspAP? tinyproxy

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
Tell us what you observed instead.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

billz, 2020-03-18:
I have not used tinyproxy so cannot advise you on this. Best of luck

closed 2020-03-18 by billz

marafa, 2020-03-19:
me bad. i was asking a question in general and not specifically about tinyproxy, i was just answer the question the question if any else was running on the pi box. and for completeness, there is also nodogspash

my question was and still is how to configure raspap with _any_ proxy. 

thank you