RaspAP/raspap-webgui #534: No internet connection in Wifi client AP mode

lemon-bean, 2020-03-21:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I'm living in a damn country where internet connection to the outside world is so weak that I couldn't finish a proper installation using your convenient one line command installer. So I installed Raspap manually following your instruction. Eventually, It runs, thank God. But after I switched it to the Wifi client AP mode, I just couldn't access the internet.

Sorry, I'm a newbie, and tried a lot to solve this problem, but I still can get a clue. So I have to ask you for help though I don't want cost your time on this small problem. 

**Your environment**
* Raspberry Pi hardware: Pi 4 B+  
* Raspbian version: Buster Desktop 10
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Manual
* Using default configuration? Y but changed the wifi Country Code to China
* Simultaneous AP and managed mode? I guess Y, though don't clearly sure what that means
* Onboard wireless chipset or external adapter? onboard wireless
* Other software or services running with RaspAP? None

**Steps to reproduce**
Sorry, I don't know. But these may help: in the "Hotspot" section, "Basic" tab, I couldn't save any changes to the "interface" option. It is fix at "uap0". and I don't know whether it is ok or not.

**Expected behavior**
Got a proper connection to the internet via Wifi client AP mode.

**Actual behavior**
No access to the internet.

**Screenshots**
[image]

[image]

**Additional context**
Could you please add some plugin software helping me escape from the firewall of my damn country? Such as V2Ray, Trojan, or ShadowsocksR plus? 

Thank you for your time and help!

billz, 2020-03-21:
@lemon-bean I'm available to help troubleshoot under these circumstances. Send me an email (address is in index.php) and we'll try to get you online.

lemon-bean, 2020-03-21:
Thank you! Sending the email now.

lemon-bean, 2020-03-23:
**Summary of Solution** for anyone encounters similar situation.

Thanks to billz, who has been kindly and patiently help me went through the whole problem solving process.

The problem was caused by a small issue during the manual installation, mainly about these three commands:

`echo 1 > /proc/sys/net/ipv4/ip_forward #RASPAP <br />`
`iptables -t nat -A POSTROUTING -j MASQUERADE <br />`
`iptables -t nat -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE #RASPAP`

When I run the first command during the installation, I got this:

`pi@raspberrypi:~ $ echo 1 > /proc/sys/net/ipv4/ip_forward`
`-bash: /proc/sys/net/ipv4/ip_forward: Permission denied`

What you should do is:

`sudo nano /proc/sys/net/ipv4/ip_forward`

Then find “#net.ipv4.ip_forward=1”, and remove the # at the beginning of the line. Then press CTRL + 'o' save the file, and CTRL + 'x' close the file. After this you can use below command to check if the file has been correctly modified.

`cat  /proc/sys/net/ipv4/ip_forward`

If it does, you can move to the next two commands, which are updating iptables for network traffic control. You may also got some error messages when you execute these two commands like me, or you may found that after manual installation, every time the Pi reboots, your iptables setting just been restored by the system. In order to fix these, you can do this:

1. Install iptables-persistent

`apt-get install iptables-persistent`

2. Start iptables-persistent service

`service iptables-persistent start`

3. Run those two commands with sudo

`sudo iptables -t nat -A POSTROUTING -j MASQUERADE #RASPAP`
`sudo iptables -t nat -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE #RASPAP`

That's all. Hope you enjoy RaspAP!

closed 2020-03-23 by lemon-bean