RaspAP/raspap-webgui #536: Option switches do not persist state on first save

billz, 2020-03-22:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
The Bootstrap custom-switch toggles on the **Hotspot** > **Advanced** tab require two form submits before they indicate their state. It's unlikely the problem stems from the controls themselves, but is rather caused by the handling/display of these states. Possibly a bug related to template rendering?

**Your environment**
* Raspberry Pi hardware: n/a
* Raspbian version: Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**
Navigate to the **Hotspot** > **Advanced** tab, change a toggle state and choose 'Save settings'. Tested with latest MacOS Chrome.

**Expected behavior**
The toggle shall indicate its desired state (on or off) after one form submit. 

**Actual behavior**
Toggle state requires two form submits before it indicates the desired state.

**Screenshots**
[image]

**Additional**
For improved UX, ajaxify these toggles so an explicit form submit is not needed. Indicate the saved state visually with a green checkmark or similar.

Taikuh, 2020-03-23:
**Another way to reproduce**

Navigate to the **Hotspot** > **Advanced** tab, change a toggle state and choose 'Save settings'. 
Now, instead of doing another form submit as indicated in your **Actual behavior**, click **Hotspot** again and navigate to the **Advanced** tab and you'll see that the toggle is correctly updated.

Actually, if we do your way of 2x form submits, then "refresh" by clicking **Hotspot** then viewing **Advanced** tab, you'll see that the toggle is back to off (the initial state) because you did in fact toggle it on (1st form submit), then back off (2nd form submit).

billz, 2020-03-23:
@Taikuh yep, you're right. I just noticed this. A second page load indicates the saved state. I was acting like a typical irritated user and re-submitting the form to make it stick ;)

On the topic of UX, it might be an idea to preserve the selected tab state when settings are saved. I find myself navigating back to the tab to check/modify the values. Little tweaks to enhance usability.

Taikuh, 2020-03-23:
Somewhat following this [SO answer](https://stackoverflow.com/a/33508530), I added
```php
echo "<meta http-equiv='refresh' content='0'>";
```
after [Line 31 in `includes/hostapd.php`](https://github.com/billz/raspap-webgui/blob/72eec3e2afbd75cbd61f1fd17302f7ccea54ebf4/includes/hostapd.php#L31).

This indeed triggers a page reload and shows the toggle in the correct state. However, the status message "Wifi Hotspot settings saved" disappears on the reload.

Sorry I can't be of more help!

>On the topic of UX, it might be an idea to preserve the selected tab state when settings are saved.

Yes, agreed!

Taikuh, 2020-03-23:
Disregarding the Line 31 changes from my previous comment, I'd like to keep my train of thought running a bit.

`index.php` reloads on POST submit, which is evident by the new status message "Wifi Hotspot settings saved" that appear upon "Save settings". However `$arrHostapdConf` isn't refreshed, so the toggles aren't displaying the correct state. This is more evident if you add to the beginning of `templates/hostapd.php`:
```php
<pre><?php print_r($arrHostapdConf) ?></pre>
```

billz, 2020-03-23:
Good catch. The toggles on the DHCP server tabs don't exhibit this problem, so maybe a clue there.

Taikuh, 2020-03-23:
Good tip! PR is a quick fix for now. Nothing extra: no ajaxified toggles, no returning to the same tab.

billz, 2020-03-23:
Fixed by @Taikuh with #539

closed 2020-03-23 by billz