RaspAP/raspap-webgui #538: No List of connected clients in Bridge Mode

stampf, 2020-03-23:
I installed RaspAP v2.2 on a fresh Raspbian Lite install (through the official installer). Everything worked out of the box (a very big thank for that!).
I switched to Bridge mode in order to be able to access stuff on my existing network (NAS, Chromecast). But now the list of connected clients is empty (was updated in Routed mode)

billz, 2020-03-23:
In bridged AP mode your Pi ceases to act like a router and forwards DHCP requests upstream. Check your router's admin UI, you should see connected clients there.

stampf, 2020-03-23:
Yes, of course I see the DHCP clients, but I expected to see the Wifi connected ones in the dashboard's lower right bloc titled "Connected devices"... which are connected to the wifi of the hotspot...

billz, 2020-03-23:
The dashboard displays dnsmasq leases compared with the system's arp cache.

https://github.com/billz/raspap-webgui/blob/cb65a66ee3b0532959b1b6b72beaa5c4d00194b7/templates/dashboard.php#L8

Your Pi may have leases from previously connected clients, but it won't display these if there isn't a match between the AP's arp table and the client's MAC.

Verify this with `sudo arp`. Compare the results in routed and bridged AP modes.

Taikuh, 2020-03-24:
@stampf, see #302 for the PR that first included the Connected Device feature. Since then, the scope of this project has grown to include the Bridged AP mode which doesn't use dnsmasq and its DHCP server functionality, resulting in an empty Connected Device list.

I, too, would like to see the list in Bridged mode and am thinking of alternate ways to populate the list and display the relevant data (Hostname / IP Address / MAC Address).

Instead of using dnsmasq leases, I'm leaning towards using iw station dump to collect MAC addresses, then cross referencing them with wlan0's arp table. Then some more awk/sed/grep magic to get the relevant data. Any thoughts, suggestions, comments are welcome.

closed 2020-03-25 by billz