RaspAP/raspap-webgui #540: Openvpn page on latest master

feh123, 2020-03-24:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I cannot open the openvpn page on the UI. If I try to do so the UI stalls and the only remedy I have found is to reboot. I can check that the VPN is working fine from my tablet using ipleaks so it's not an issue with the VPN itself.

**Your environment**
* Raspberry Pi hardware: Pi 3 Model B  
* Raspbian version: Buster Lite
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y)
* Onboard wireless chipset or external adapter? 
* Other software or services running with RaspAP? None

**Steps to reproduce**
Setup an Openvpn - mine has run successfully for the last six months. Try to view Openvpn page.

**Expected behavior**
I expect to be able to open the page and to see my VPN IP address.

**Actual behavior**
Cannot open this page. So if I am on Dashboard and try to open the VPN the UI stalls. I have not tried systematically opening the VPN from other pages.

**Additional context**
My AP has been working well for +six months. Only recently (not the very last master) has it shown this behaviour. I thought it maybe fixed as more new developments were added. But updated today and issue still present. Not a major problem but it would be good to know my IP address from the UI for convenience. Thanks.

Taikuh, 2020-03-25:
Hi, I've tried but couldn't reproduce your problem. Have you tried a fresh install of Raspbian and Raspap? Below is a quite verbose step-by-step (a lot dedicated to setting up simultaneous AP and managed mode):

- unplug ethernet cord from Pi
- fresh Rasbian Buster Lite install > apt update and upgrade and reboot
- use raspap quick installer and reboot
- on my laptop, connect to raspap-webgui
- access webgui at 10.3.141.1 > WiFi client > connect to upstream router's wifi
- Error msg: Wifi settings updated but cannot restart (cannot execute "wpa_cli reconfigure")
- follow [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#wifi-scanning-doesnt-work-or-i-get-the-error-cannot-execute-wpa_cli-reconfigure-help) > raspap-webgui AP subsequently disabled
- on my laptop, connect to upstream router's wifi
- access webgui at `hostname`.local > WiFi client > connect to same upstream router's wifi > success
- Hotspot > Advanced tab > Wifi Client AP mode > Save settings > Restart Hotspot
- webgui hangs > reboot pi
- on my laptop, connect to raspap-webgui
- access webgui at 192.168.50.1 > OpenVPN > enter creds and upload ovpn > Save settings > Start openvpn
- webgui doesn't hang, can confirm IP changes to VPN's

billz, 2020-03-25:
Hmm, that's a weird one. I'm also unable to repro with latest master. 
Try giving lighttpd a kick `sudo systemctl restart lighttpd.service`.
This won't affect OpenVPN or other running services.

I pushed a commit yesterday to fix some minor php notices. Not specific to this issue, but you could do a fresh pull of the master branch.

Taikuh, 2020-03-25:
To add, if @feh123 is upgrading RaspAP as per the [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#upgrade) instead of doing fresh installs to upgrade, that could be a reason for his breaking.

From a usability point of view, a more fool-proof way of upgrading while keeping user config settings, could make for a good mid-to-long term project.

feh123, 2020-03-25:
Thanks for both your responses - much appreciated. I have re-installed from fresh, activated the managed mode and set up my VPN. I can enter the VPN page without an issue. Everything looks good. So maybe the upgrade route has an issue as you suggest. Thanks for the tip about kicking lighttpd.

closed 2020-03-25 by feh123

billz, 2020-03-25:
> a more fool-proof way of upgrading while keeping user config settings, could make for a good mid-to-long term project

@Taikuh indeed. this came to light in my rounds of testing Armbian/Debian.

Along these lines, the installer currently indicates the latest release version, but pulls from master. The master branch is always 'production ready', at least in theory. However there could be a case for prompting the user at install time if they want the most recent stable release or the latest 'bleeding edge' commits. Anyway, probably grounds for a new issue.