RaspAP/raspap-webgui #542: Blocking ads, trackers & other online garbage

billz, 2020-03-27:
I've heard from users on lockdown due to the pandemic who are struggling with slow internet. Seems to me this could help. Looking at the dnsmasq [docs](http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html):

```
-H, --addn-hosts=<file>    Additional hosts file. Read the specified file as well as
                           /etc/hosts. If -h is given, read only the specified file.
                           This option may be repeated for more than one additional
                           hosts file. If a directory is given, then read all the files
                           contained in that directory.
```
This option could be used to create an ad / tracker blocking option, effectively ignoring DNS requests for undesirable hosts. 

If it sounds familiar, this is indeed very similar to Pi-Hole. However, combining this facility with an AP would simplify setup and improve throughput by filtering out cruft, at least in theory.

[This project](https://github.com/notracking/hosts-blocklists ) looked promising, so I quickly prototyped a new feature that uses it. There are of course [many other such lists](https://github.com/StevenBlack/hosts).

[image]

The update button pulls the latest lists down to the Pi. I haven't done extensive testing (ad blocking, performance or otherwise), but dnsmasq seems happy with it.

```
dnsmasq[20285]: read /etc/raspap/adblock/hostnames.txt - 78506 addresses
```

Anyone have experience with this?

billz, 2020-03-27:
Ad blocking confirmed 🛑 :) 

```
dnsmasq[2348]: query[A] googleads.g.doubleclick.net from 10.3.141.160
dnsmasq[2348]: config googleads.g.doubleclick.net is 0.0.0.0
dnsmasq[2348]: config c.o0bc.com is 0.0.0.0
dnsmasq[2348]: config realestate.w.o0bc.com is 0.0.0.0
dnsmasq[2348]: config c.amazon-adsystem.com is 0.0.0.0
dnsmasq[2348]: config cdn.krxd.net is 0.0.0.0
dnsmasq[2348]: query[A] i.clean.gg from 10.3.141.160
dnsmasq[2348]: /etc/raspap/adblock/hostnames.txt i.clean.gg is 0.0.0.0
dnsmasq[2348]: config ib.adnxs.com is 0.0.0.0
dnsmasq[2348]: query[A] run.adrizer.com from 10.3.141.160
dnsmasq[2348]: /etc/raspap/adblock/hostnames.txt run.adrizer.com is 0.0.0.0
dnsmasq[2348]: query[A] securepubads.g.doubleclick.net from 10.3.141.160
dnsmasq[2348]: config securepubads.g.doubleclick.net is 0.0.0.0
dnsmasq[2348]: query[A] cdn-gl.imrworldwide.com from 10.3.141.160
dnsmasq[2348]: config cdn-gl.imrworldwide.com is 0.0.0.0
dnsmasq[2348]: query[A] audit.quantcast.mgr.consensu.org from 10.3.141.160
dnsmasq[2348]: /etc/raspap/adblock/hostnames.txt audit.quantcast.mgr.consensu.org is 0.0.0.0
dnsmasq[2348]: query[A] sb.scorecardresearch.com from 10.3.141.160
dnsmasq[2348]: config sb.scorecardresearch.com is 0.0.0.0
dnsmasq[2348]: query[A] app-measurement.com from 10.3.141.160
dnsmasq[2348]: config app-measurement.com is 0.0.0.0
dnsmasq[2348]: query[A] sdk.iad-01.braze.com from 10.3.141.160
dnsmasq[2348]: config sdk.iad-01.braze.com is 0.0.0.0
```

billz, 2020-04-01:
Ad blocking installs cleanly on Raspbian and Armbian. Tested against several commercial sites with the aim of collecting baseline data. Have not gone so far as to measure throughput, but site rendering is noticeably improved. Trackers are blocked as well.

Sample logs here https://pastebin.com/raw/caieK2vU

Don't ask me what summerhamster.com is. Anyway, good news is it's blocked :)  

I'll post details for beta testers shortly.

billz, 2020-04-03:
PR merged and released as beta. Details [on the wiki](https://github.com/billz/raspap-webgui/wiki/Ad-blocking-(Beta))

marafa, 2020-04-03:
i have configured raspap-gui to use [nxfilter](https://nxfilter.org)
- bridged mode
- set the dns to use the nxfilter ip

the advantage of nxfilter is obvious when it you have kids in virtual schools at home . this [video]( https://www.youtube.com/watch?v=wd1-TOuuiYo) explains how to configure nxfilter for that purpose

i intend to write up a doc in a PR for this over the weekend

billz, 2020-04-03:
@marafa sounds great. looking forward to your write up

firestrife23, 2020-04-04:
I thought I should share this for those who might be interested in using Adguard Home. I found an easier workaround with Adguard Home running along with RaspAP. All I had to do to change Adguard Home’s listening port to 5300 and bind to 127.0.0.1, then go to RaspAP > DHCP Server > Advanced page and turn on “Upstream DNS Server” and add 127.0.0.1#5300 as an upstream DNS Server.

billz, 2020-04-05:
@firestrife23 good tip, thanks. I'll add this to the FAQ

billz, 2020-04-13:
Ad blocking is an installable option in the latest 2.4 release.

HoM3r17, 2020-04-25:
> Ad blocking is an installable option in the latest 2.4 release.

Is there a way to enable it afterwards? i've installed 2.4 a few days ago and didn't thing i would need it at this time ...

billz, 2020-04-25:
Either run the installer again or use the [manual installation](https://github.com/billz/raspap-webgui/wiki/Manual-installation). You can just follow the last steps at the end specific to ad blocking.

> (Optional) Install Ad blocking, enabling the option in RaspAP's config

ShibaeO, 2020-05-31:
hi, 

can i still use ad block in briged mode, or anything else wich can do the same work ?

thx

billz, 2020-05-31:
> can i still use ad block in bridged mode, or anything else which can do the same work ?

In bridged mode DNS is handled by your router, so ad blocking is disabled in RaspAP. 

> can i install pi-hole be-side raspap ?

Short answer is no, because both projects use `dnsmasq`. See [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#pihole) for more info. You can however run [Adguard Home on the same Pi](https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-integrate-raspap-with-adguard-home).

ShibaeO, 2020-06-01:
hi thank you for you quick anwser :)

my real objective is to block ad in the youtube app of my ipad and and my phone but i tried the adguard public dns and it dont block them, so im gonna keep my lovely ads to support creator ^^
i prefer have hombridge working :)

but btw nice project thx for you work

closed 2020-09-08 by billz

GCleuziou, 2021-02-18:
> i have configured raspap-gui to use [nxfilter](https://nxfilter.org)
>
>     * bridged mode
>
>     * set the dns to use the nxfilter ip
>
>
> the advantage of nxfilter is obvious when it you have kids in [...]

Hi @marafa,
I'm interested in your solution. Actually I'm really a beginner but I'm trying to make RaspAP and NxFilter cohabit on my raspberry. Each works well independently but I have a port conflict problem when I try to make them work together. I see that you are using bridget mode on RaspAP but your access point is no longer usable (DHCP router not working with this mode, isn't it?)? I want my clients to be managed on an address range defined by RaspAP...

marafa, 2021-02-19:
https://github.com/RaspAP/raspap-webgui/pull/548

confever, 2023-10-14:
> I thought I should share this for those who might be interested in using Adguard Home. I found an easier workaround with Adguard Home running along with RaspAP. All I had to do to change Adguard [...]

hello, if i may ask. which one should i install first ? adguard then raspap, or raspap then adguard ?
and if my pi's static ip is at 198.123.3.1, would raspap and adguard use the same static ip for admin panel ?