RaspAP/raspap-webgui #544: bridged AP mode not working

Langelot56, 2020-03-29:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [x ] I checked the FAQ before creating this issue
* [x ] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

**Your environment**
* Raspberry Pi hardware: Pi 4B 
* Raspbian version: Raspbian Buster with desktop and recommended software
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? Onboard chipset
* Other software or services running with RaspAP? No, just fresh install

**Steps to reproduce**
I have a problem with the bridged AP mode. As I used it on an install with other software installed, I decided to make a fresh install of raspbian to confirm the problem. 
I have: 
-followed the prerequisites
-installed raspap with "curl -sL https://install.raspap.com | bash" and said yes to all questions
-rebooted
-connected to raspap configuration portal and slided the bridged AP mode
-rebooted

**Expected behavior**
I expected to access the raspberry from an other device on the local network.

**Actual behavior**
Unable to reach the raspberry from an other computer on the network.
To be more accurate:
-before toogling bridged AP mode, I had access (raspberrypi.local) and I could see the hostname "raspberrypi" on my router page showing connected devices (through ethernet)
-after it was no longer possible
-I toogled to normal mode again, access and presence on the router page came back again.

**Screenshots**

**Additional context**

I identified it as a bug. Do you have any idea on this issue?
Thank you!

billz, 2020-03-29:
did you [read this](https://github.com/billz/raspap-webgui/wiki/Bridged-AP-mode)? 
is the bridged AP visible? able to connect?

home routers differ vastly in terms of capabilities. some resolve the new bridged client IP almost immediately, others can take longer. some devices, including the Pi, may require a restart with bridged AP mode enabled.

Langelot56, 2020-03-30:
Thank you for your answer. Yes I read this page before asking for help. Restarting the Pi with bridge AP mode enabled and restarting the router has no effect, (same behaviour).
This command " ifconfig br0 | grep 'inet ' | awk '{print $2}' " returns indeed an IP address. However I can't access the Pi with this ip address from another computer on the local network... And the Pi still does not appear on the connected devices list of the router.
The bridge mode is indeed activated (checked by connecting on the AP interface directly on the Pi).

billz, 2020-03-30:
I don't like saying it works for me, but... it works for me ;) 

did you reboot your pi? beyond this I can't help you troubleshoot your network. perhaps @Taikuh or others have some suggestions.

Langelot56, 2020-03-30:
Ok, thank you for your help anyway! Yes, I reboot the pi and the router several times. As I did everything by the book and as it works on your side, I guess that the issue comes from my router... You were able to test with the same config (Pi 4B) ?

closed 2020-05-15 by stale[bot]

ProjektFEL, 2021-01-12:
Hello, i have same issue with RPI 3B+, after toggling to bridge AP, device wouldnt start correct. i have also HDMI to see desktop and there is no signal... i tried it two times and its same. Do you have some specific version which works on RPI 3B+ in bridge AP mode?

bytemaster0, 2021-12-09:
Have the same problem. Running a Ubiquiti EdgeRouter 4 (VyOS) as my router, everything hard-wired. WiFi devices don't get an IP, and eth0 is not accessible at all from the network.