RaspAP/raspap-webgui #546: Does default install require a reboot?

OneOfTheInfiniteMonkeys, 2020-03-30:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
RaspAp does not start following default install on new system. No request or instruction to restart system.

**Your environment**
* Raspberry Pi Zero W Rev 1.1
* Raspbian GNU/Linux 10 (buster) Linux 4.19.97+
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install CURL command line
* Using default configuration? (N)
  Install to web folder, elected to install web to /var/www/html/raspap
* Simultaneous AP and managed mode? (Y/N)
  Not possible as system not running, however is the goal
* Onboard wireless chipset
* Other software or services running with RaspAP?
  None - Clean install test

**Steps to reproduce**
  In P.C:
  Format 64 Gb Samsung Evo Plus MicroSD card with Fat32  using fat32format.exe
  Copy NOOBS_lite_v3_3 to MicroSD card
  Create empty ssh file
  Create wpa_supplicant.conf with WiFi configration parameters:
    ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
    update_config=1
    country=GB
    network={
      ssid="<VALID ACCESSPOINT NAME REMOVED>"
      psk="<VALID PASSWORD REMOVED>"
    }
  Place MicroSD card in PiZero W and power up:
  Perform Noobs Lite install.
  Change root password
  sudo apt-get update && apt-get upgrade
  sudo apt-get dist-upgrade. 
  Reboot PiZero W. 
  Check network connectovity by using remote SSH
  Use quick install CURL command to install RaspAp. 
  Accept all defaults apart from default web install location for RaspAp Lighttpd location. 
  Change web app location for Lighttpd to /var/www/html/raspap
  Attepmt to connect to WiFi hotspot (not active)
  Attempt to connec to to lighttpd via Access Point assigned address (no lighttpd)

**Expected behavior**
A clear and concise description of what you expected to happen.
Request for reboot if required 
or
WiFi hotspot starts, Web configuration service available at either WiFi or Access Point assigned IP address.

**Actual behavior**
Tell us what you observed instead.
No WiFi Hotspot, no web configuration services at either WiFi (no hotspot) or Access Point assigned IP address.

**Screenshots**
If applicable, add screenshots to help explain your problem.
None - see htop output below.

**Additional context**
Add any other context about the problem here.
htop 

Tasks:  83 total,   1 running,  82 sleeping,   0 stopped,   0 zombie
%Cpu(s): 13.0 us, 21.7 sy,  0.0 ni, 65.2 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :    432.4 total,     98.4 free,     42.6 used,    291.4 buff/cache
MiB Swap:    100.0 total,     98.7 free,      1.2 used.    330.3 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
10664 pi        20   0   10312   2992   2488 R   5.0   0.7   0:00.10 top
    1 root      20   0   33748   8228   6488 S   0.0   1.9   1:27.57 systemd
    2 root      20   0       0      0      0 S   0.0   0.0   0:00.01 kthreadd
    6 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mm_percpu_wq
    7 root      20   0       0      0      0 S   0.0   0.0   0:03.69 ksoftirqd/0
    8 root      20   0       0      0      0 S   0.0   0.0   0:00.01 kdevtmpfs
    9 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 netns
   11 root      20   0       0      0      0 S   0.0   0.0   0:00.01 khungtaskd
   12 root      20   0       0      0      0 S   0.0   0.0   0:00.00 oom_reaper
   13 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 writeback
   14 root      20   0       0      0      0 S   0.0   0.0   0:00.00 kcompactd0
   15 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 crypto
   16 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kblockd
   17 root      rt   0       0      0      0 S   0.0   0.0   0:00.00 watchdogd
   18 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rpciod
   20 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 xprtiod
   23 root      20   0       0      0      0 S   0.0   0.0   0:00.60 kswapd0
   24 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 nfsiod
   35 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kthrotld
   36 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 iscsi_eh
   37 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 dwc_otg
   38 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 DWC Notificatio
   39 root       1 -19       0      0      0 S   0.0   0.0   0:00.00 vchiq-slot/0
   40 root       1 -19       0      0      0 S   0.0   0.0   0:00.00 vchiq-recy/0
   41 root       0 -20       0      0      0 S   0.0   0.0   0:00.00 vchiq-sync/0
   42 root      20   0       0      0      0 S   0.0   0.0   0:00.00 vchiq-keep/0
   43 root      10 -10       0      0      0 S   0.0   0.0   0:00.00 SMIO
   45 root     -51   0       0      0      0 S   0.0   0.0   0:05.31 irq/86-mmc1
   48 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mmc_complete
   49 root       0 -20       0      0      0 I   0.0   0.0   0:04.18 kworker/0:1H-kblockd
   51 root      20   0       0      0      0 S   0.0   0.0   0:00.83 jbd2/mmcblk0p7-
   52 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 ext4-rsv-conver
   53 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 ipv6_addrconf
   90 root      20   0   15940   6800   5880 S   0.0   1.5   0:03.32 systemd-journal
  109 root      20   0   17944   3832   3060 S   0.0   0.9   0:01.44 systemd-udevd
  120 root      10 -10       0      0      0 S   0.0   0.0   0:00.00 SMIO
  127 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mmal-vchiq
  129 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mmal-vchiq
  130 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mmal-vchiq
  149 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 cfg80211
  150 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 brcmf_wq/mmc1:0
  151 root      20   0       0      0      0 S   0.0   0.0   0:01.06 brcmf_wdog/mmc1
  200 systemd+  20   0   22396   5588   4956 S   0.0   1.3   0:01.10 systemd-timesyn
  235 root      20   0   13132   5768   5088 S   0.0   1.3   0:07.11 systemd-logind
  240 nobody    20   0    4304   2096   1924 S   0.0   0.5   0:00.36 thd
  245 avahi     20   0    5888   2832   2492 S   0.0   0.6   0:12.76 avahi-daemon
  252 root      20   0    7960   2340   2136 S   0.0   0.5   0:00.10 cron
  259 avahi     20   0    5756   1520   1296 S   0.0   0.3   0:00.00 avahi-daemon
  263 message+  20   0    6684   3640   3076 S   0.0   0.8   0:18.81 dbus-daemon
  270 root      20   0   10724   4456   4052 S   0.0   1.0   0:00.30 wpa_supplicant
  272 root      20   0   24604   3488   2744 S   0.0   0.8   0:00.54 rsyslogd

billz, 2020-03-30:
> Does default install require a reboot?

Yes. Per the installer:

> The system needs to be rebooted as a final step. Reboot now?

OneOfTheInfiniteMonkeys, 2020-03-31:
I apologise, I must have missed that instruction.

If I may I recommend the "Quick installer" instructions on the RaspAp home page perhaps be ameded to include "reboot required".

closed 2020-03-31 by OneOfTheInfiniteMonkeys