RaspAP/raspap-webgui #552: RPI 3B Wlan0 as AP and Client not working

feh123, 2020-04-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [x ] I checked the FAQ before creating this issue
* [x ] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

**Your environment**
* Raspberry Pi hardware: 3B  
* Raspbian version: Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y. Openvpn activated.
* Onboard wireless chipset or external adapter? Onboard 
* Other software or services running with RaspAP? None

**Steps to reproduce**
Successfully setup AP then after a few hours AP drops connection to the Internet.

**Expected behavior**
AP is stable.

**Actual behavior**
Loss of internet.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I still have problems running AP/managed mode on a RPI 3B with openvpn active - with all the versions from sometime after the addition of OpenVPN (cannot pin down the exact date) to the latest containg Ad block. Several times I have started from a fresh install and always I see the same issue. Although successful initially after a few hours connected devices lose the wifi.  SSID is still there and I can connect to it but then the device has no internet.
At first I thought it maybe due to updating rather than installing from fresh. But now I have tried from fresh and still get the same issue. Does anyone see this issue too? Thanks.

billz, 2020-04-06:
> Provide useful detail to reproduce your issue. "Doesn't work" or "not working" is not a valid report.

The hostapd logs are there for a reason ;)

Do you observe this without openvpn active?
Have you checked for errors with your openvpn client?
`journalctl | grep openvpn`

feh123, 2020-04-06:
Sorry - I am not sure it is a bug and question is a better label. I tried to say what happened in the additional info section. I have not looked in the logs - the problem is that the RPI also disconnects from my network. The SSID is still broadcasting but I can no longer ssh into the RPI and the UI does not work. 
I have checked my VPN and made sure I was up to date but the RPI still disconnects. I have another RPI running Buster but using eth0 and the same VPN provider - it is working fine. 
I have not tried it without the vpn - as this seem to be working fine I thought problem was elsewhere. My main thought was that perhaps others are also seeing this - but if not then there is something wrong with my set up. I will install again and look at the logs without and with the vpn.

billz, 2020-04-06:
No need to apologize. Descriptive bug reports backed with logs/other data help us improve this project.

feh123, 2020-04-07:
I have re-installed from fresh with the adblock version and a new sd card. I activated managed mode but did not use the openvpn. There is some improvement - the AP works although is much slower than normal. I then activated the VPN and the AP is still working. However the UI crashed after trying to check the adblock page. Before it crashed when I tried to look at the openvpn page so I am not sure the page is significant. As before when the UI goes the RPI is no longer on the network yet the AP is still present. I can restore the UI by unplugging. Re logs - I had enabled logging and I could see no errors. I have also tried journalctl | grep openvpn and that also shows no errors. 
I have a second RPI running eth0/wlan0 with the VPN and that is working fine.

closed 2020-05-15 by stale[bot]