RaspAP/raspap-webgui #557: Changing IP-Address

Michael-S-de, 2020-04-09:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [x ] I checked the FAQ before creating this issue
* [x ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I am not able to change the IP-Address of the AP

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  Pi 3B+
* Raspbian version (examples: Buster Lite, Buster Desktop) Buster Desktop
* Followed the project prerequisites? (Y/N) Y
* Checked the project FAQ? (Y/N) Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y/N) see description
* Simultaneous AP and managed mode? (Y/N) simultaneous AP
* Onboard wireless chipset or external adapter?  onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.
A clear and concise description of what the bug is.
I am developing a system with an Android Phone as AP. During development I like to use the Raspberry Pi instead. (Connected to Lan ) Android APs has the hard wired IP 192.168.43.1
I installed RASPAP on a new Buster and Test it: Everythink ok. I connect a Phone and get Internet connection. I was able to change SSID and Passwort
Then I tried to change the IP-Adress with the help of this:
https://blog.sebastian-martens.de/technology/creating-wifi-access-point-with-raspberry-pi-using-raspap
Result: SSID is visible but connection fails with "no internet"
Your configuration tool shows under network/wlan0 static IP/subnet NaN.NaN.NaN.NaN
Your programm changed /etc/dhpccd.conf static ip_address=192.168.43.1/24 to .../INF
Each time, when I try to change this information, the programm sets it back.
Will I need additional settings?
**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
Tell us what you observed instead.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

billz, 2020-04-09:
The guide you followed is outdated. This project no longer modifies `/etc/dnsmasq.conf` directly, so you probably have an IP conflict. Try this instead.

https://github.com/billz/raspap-webgui/blob/223ae3e11ad73b70e5c70fbce31227f1e3324ddf/config/config.php#L12

Michael-S-de, 2020-04-13:
Thank you for the quick answer.

The content of /etc/dnsmasq.d/090_raspap.conf is:

interface=wlan0

dhcp-range=192.168.43.50, 192.168.43.249,255.255.255.0,12h

log-facility=/tmp/dnsmasq.log

conf-dir=/etc/dnsmasq.d

I cannot understand what should be wrong with this.

The RASPAP-App still shows network/wlan0 Subnet NaN.NaN.NaN.NaN. The static IP should be 192.168.43.1. Gateway same IP (See picture)

The DHCP settings are 192.168.43.50 to 192.168.43.249

See that picture.

Any advice?

https://github.com/billz/raspap-webgui/blob/223ae3e11ad73b70e5c70fbce31227f1e3324ddf/config/config.php#L12

billz, 2020-04-17:
Essentially a duplicate of #329. Refer to [this comment](https://github.com/billz/raspap-webgui/issues/329#issuecomment-517689661) in particular.

tl;dr: this project does not merge manual configs done outside of the app.

closed 2020-06-23 by stale[bot]