RaspAP/raspap-webgui #563: [Feature-Request] User Accounts like Hotspot (like Telekom)

zandercodes, 2020-04-13:
Is it possible to add user accounts?

Explanation: I would like to offer a free wlan but it should be limited by accounts.
The user should first register on the login page, e.g. at Telekom Hotspot, and then be able to use the internet.

Best regards
ZanderCodes

billz, 2020-04-14:
Something [like this](https://github.com/billz/raspap-webgui/wiki/FAQs#captive)?

zandercodes, 2020-04-21:
Yes something like that but with user registration or can you do that with it?

billz, 2020-04-22:
Per the [NDS wiki](https://github.com/nodogsplash/nodogsplash/wiki/1-What-is-NoDogSplash%3F#1-overview):

> If you want to link Nodogsplash to your own centralised Internet based authentication service with user account self generation and access charging, you can do that too, or anything in between.

This is within project scope, but not a priority in the foreseeable future. 
PR's are welcome :)

zandercodes, 2020-04-24:
Thank you.
It's all resolved.

closed 2020-04-24 by zandercodes