RaspAP/raspap-webgui #564: Reinstalling raspap by script, lighttpd.conf is not restored

lewixlabs, 2020-04-16:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [] This is a bug report
* [x ] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the help**
Probably is not a bug, but I ask you to help me to restore 
/etc/lighttpd/lighttpd.conf
because after uninstalling raspap & lighttpd, this file does'nt exist anymore in this path

Trying to re-laungh script
curl -sL https://install.raspap.com | bash
/etc/lighttpd/lighttpd.conf is not restored (still not found)
so after a complete raspap install, lighttpd doesn't start because doesn't find a valid config file.

Thanks to anybody can help me, maybe could you send to this post the default content of lighttpd.conf file set by raspap?

**My environment**
* Raspberry: Pi Zero
* Raspbian version: Buster Desktop (examples: Buster Lite, Buster Desktop)
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Yes
* Simultaneous AP and managed mode? No
* Onboard wireless chipset or external adapter? Onboard 
* Other software or services running with RaspAP? No

billz, 2020-04-16:
Doesn't sounds like a bug since the uninstaller performed its function. But anyway, here's the [default lighttpd.conf ](https://gist.github.com/billz/63913bd364c24af1f8034db24dd9fc7f)from my RPi. Hope this helps.

lewixlabs, 2020-04-17:
Thank you @billz !

> uninstaller performed its function (uninstalling lighttpd and deleting lighttpd.conf)
Yes, I agree, but it is weird relaungchin installer (curl -sL https://install.raspap.com | bash)
/etc/lighttpd/lighttpd.conf is not set

Now I relaunched
curl -sL https://install.raspap.com | bash
from scratch from my rpi zero
I put your lighttpd.conf content, chaning port only because on port 80 I have AdGuard Home.

Now logs show me these errors:

root@raspberrypi:~# tail -f /var/log/lighttpd/error.log
2020-04-16 11:19:40: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: page in /var/www/html/index.php on line 46
2020-04-16 11:19:40: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: sidebarToggled in /var/www/html/index.php on line 55
2020-04-16 11:19:40: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: File o directory non esistente in /var/www/html/index.php on line 60
2020-04-16 11:19:40: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: File o directory non esistente in /var/www/html/templates/dashboard.php on line 2
2020-04-16 11:19:40: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: page in /var/www/html/templates/dashboard.php on line 134
2020-04-16 11:21:45: (server.c.2059) server stopped by UID = 0 PID = 1 
2020-04-16 17:09:23: (server.c.1464) server started (lighttpd/1.4.53) 
2020-04-16 17:09:23: (server.c.1493) WARNING: unknown config-key: fastcgi.server (ignored) 
2020-04-16 17:09:43: (server.c.958) [note] graceful shutdown started 
2020-04-16 17:09:45: (server.c.2059) server stopped by UID = 0 PID = 0 

seems hostapd.ini doesn't exist in path
/etc/raspap/hostapd.ini

billz, 2020-04-17:
> I put your lighttpd.conf content, changing port only because on port 80 I have AdGuard Home

If installing from scratch you shouldn't need to touch lighttpd.conf. You can also change the default web port on the System > Advanced tab.

lewixlabs, 2020-04-19:
I wasn't able to go to raspap gui due port 80 was used by AdGuard, (so I wasn't able to go to System->Advanced Tab.

I solved following manual installation, steb by step.

Thanks anyway!

closed 2020-04-19 by lewixlabs