RaspAP/raspap-webgui #566: Loss of connection in Bridged Mode after a while

Joy26300, 2020-04-20:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [X] I searched existing issues before opening this one
* [X] I checked the FAQ before creating this issue
* [X] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

After reboot my raspberry, generally after 1 or 2 days, the AP doesn't work, I can't connect to AP, and the password is asking for back but doesn't work.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W) : Pi 4 Model B
* Raspbian version (examples: Buster Lite, Buster Desktop) : Buster Lite 2020-02-13
* Followed the project prerequisites? (Y/N) : Y
* Checked the project FAQ? (Y/N) : Y
* RaspAP Quick Install or Manual setup? : RaspAP Quick Install v2.3.1
* Using default configuration? (Y/N) : Y
* Simultaneous AP and managed mode? (Y/N) : N
* Onboard wireless chipset or external adapter? : Onboard wireless chipset
* Other software or services running with RaspAP? : No

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

Configure Raspberry and use normally

**Expected behavior**
A clear and concise description of what you expected to happen.

AP with bridged mode, and work as router mode.

**Actual behavior**
Tell us what you observed instead.

The AP request password, but not accept good password, and obligation to reboot raspberry for work again.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

shinigami-78, 2020-04-20:
I am seeing a similar issue but right from the beginning. See the issue I just created.

billz, 2020-04-21:
@Joy26300 sounds like your Pi's DHCP lease is expiring. Check your lease status with `sudo dhclient -v` and adjust your router's settings accordingly. Many consumer-grade routers have crappy DHCP implementations, so I can't advise you there.

Joy26300, 2020-05-19:
Thanks for you're help ! He work with no problem

closed 2020-05-19 by Joy26300

asnchzv, 2020-07-13:
Hi, I have the same issue. But my internet connection is provided over ppp interface by 3G/LTE module, how can I set the configuration?

dhclient -v output:

`Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/ba:08:cd:1e:9a:e1
Sending on   LPF/wlan0/ba:08:cd:1e:9a:e1
Listening on LPF/eth0/04:d9:f5:55:76:17
Sending on   LPF/eth0/04:d9:f5:55:76:17
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 5
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 9
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 19
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 13`