RaspAP/raspap-webgui #567: Clean install on Raspberry Pi 4B gives “incorrect password error

shinigami-78, 2020-04-20:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I did a clean install on a Raspberry Pi 4B using the quick installer on the latest Raspbian release. After rebooting (post install), I connected the Raspberry Pi to my wireless network and turned on simultaneous AP and wifi client via the advanced tab on Configure Hotspot. I tried connecting to the default SSID and default password “ChangeMe” but I kept getting an incorrect password error. I tried renaming the SSID and changing the password. With that change, I could sometimes get one device to connect to Raspap but, if I tried a second device, the second device would give the same error “incorrect password” and then the first device too would fail. I tried rebooting the Raspberry Pi, which would then repeat the same behavior of sometimes allowing one device to connect and returning “password error” to any subsequent devices. I know I was entering the password correctly in every device.

**Your environment**
Raspberry Pi 4B (4gig RAM)
Raspbian Buster (latest using their Raspberry Pi Imager)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y)
* Onboard wireless chipset or external adapter? Onboard wireless chipset 
* Other software or services running with RaspAP? No

**Steps to reproduce**
Clean install of Raspbian Buster on an SD card
Clean install of Raspap using the quick install
Connect the Raspberry Pi to the router via Ethernet
Try to log into the default SSID and password; it fails
Connect to the admin panel on the Raspberry Pi via the router assigned IP (by connecting the Pi via Ethernet to the router)
Change the SSID and password
Reboot the Raspberry Pi
Connect to the new SSID using the password; one device will sometimes succeed but any second device will fail, which will cause the first device to disconnect too

**Expected behavior**
Connecting to the default SSID with the default password should work. Same thing with your chosen SSID and password.

**Actual behavior**
See above

**Screenshots**
N/A

**Additional context**
N/A

shinigami-78, 2020-04-21:
I can verify this problem does not happen with a Raspberry Pi 3B. I just did a clean install and it worked with the prior model of the Pi. The issue seems to be specific to the 4B.

sapher, 2020-04-23:
I did a clean install on a pi4 aswell, I didn't ran into this issue. I ran into other issue, like unable to have simultaneous wifi station and AP at the same time.

billz, 2020-04-28:
@shinigami-78 what do you see in the hostapd log?

billz, 2020-05-02:
AP-STA mode was [recently tested on the 4B](https://github.com/billz/raspap-webgui/pull/523#issuecomment-598802431) and found to have no issues.
If hostapd log data to the contrary is available we can regression test this feature.
Otherwise this issue will be closed as non-reproducable.

shinigami-78, 2020-05-04:
Let me do a fresh install and try again. When I ran into the problem, I moved back to the 3B and got it working; I will reflash a new microSD card tomorrow and try again with the 4B.

svenlimbach, 2020-06-12:
I see something similar on my RPI4 with RaspAP. After a while an already connected device try to reconnect and then see the wrong Password Error. The initial setup / change of SSID and Password / Connect to Clients (iPhone/iPad) worked without any problems.

svenlimbach, 2020-06-12:
> I see something similar on my RPI4 with RaspAP. After a while an already connected device try to reconnect and then see the wrong Password Error. The initial setup / change of SSID and Password / Connect to Clients (iPhone/iPad) worked without any problems.

A reboot make the connect possible again. no error message in the hostapd log

closed 2020-08-18 by stale[bot]