RaspAP/raspap-webgui #571: Make an option to have eth0 on the same DHCP than wlan0

mathojojo, 2020-04-29:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [X ] I searched existing issues before opening this one
* [X ] I checked the FAQ before creating this issue
* [X ] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

**Your environment**
* Raspberry Pi  3 Model B+
* Raspbian buster lite (updated)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? I don't know. I followed this tutorial https://docs.sixfab.com/docs/using-raspberry-pi-base-shield-as-a-wi-fi-hotspot
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (N)
* Onboard wireless chipset 
* Other software or services running with RaspAP? QMI with a Quectel EC25-E LTE Module

**Steps to reproduce**
Nothing special, just install, and see the behaviour of eth0.

**Expected behavior**

There is a dhcp server for the wireless AP (wlan0). Each WIFI device that connects to RaspAP will receive an IP in the predefined range.
I would like that if I plug a wired device on eth0, RaspAP DHCP server will attribute an IP adresse to it, in the same range.

Example : RaspAP IP : 10.3.141.1
                 Wifi device 1 : 10.3.141.10
                 Wifi device 2 : 10.3.141.11
                 Wired device  : 10.3.141.12

Most ISP modem/router are doing no difference if the connection is WIFI or wired.... there is only one DHCP server, and Wifi and wired devices are on the same network.

I'm not expert, but I think that eth0 and wlan0 should be bridged together (I tested the Bridge Mode, but it doesn't do what I hoped)

**Actual behavior**
The RaspAP Eth0 is set as a DHCP client (not a server). So if you plug a computer in this interface (the computer ethernet interface is most of the time in DHCP client mode too), nothing happen. The RaspAP won't act as an access point for this computer.It would be great if RASPAP is able to be a wired AP as well.

My case is a little more complex, but need the same Eth0 on global DHCP server : 
I have a Mesh router (Netgear Orbi system). This router has a WAN ethernet interface. It's supposed to be connected to a modem which would have a Ethernet interface for internet.
My goal was to dismiss my ISP modem /router, because it has poor features (especially OpenVPN client is missing), and use the RaspAP instead.
But the Orbi WAN input is acting like a DHCP client. It's waiting for an IP adresse from a DHCP server, DNS, gateway.

I'm really sorry if it is already possible and I didn't found the information. Please tell me if there is a way to do that, or if you have to code it.

Thank you very much for the help.

billz, 2020-04-30:
This is discussed in #324

MariusVB, 2020-06-19:
Having this feature built-in would be greatly appreciated. I can't seem to get DHCP working for both WLAN and Eth.
I'm creating an IoT system, where the AP will be created by a Raspberry Pi using RaspAP. A Windows machine with more processing power will run the back-end itself, and hence it is critical that it is always connected. To ensure this, I want to connect them by ethernet cable.

closed 2020-08-25 by stale[bot]