RaspAP/raspap-webgui #574: Backing up a configuration, and/or logging of all changes in files

frranck, 2020-05-04:
Hello Bill,
Is there a way to back up a configuration or to see all the changes being made from the web interface?
It is a great project, and it does save a lot time for bootstrapping an AP, but I had "it stopped working I have no clue what happened, let's reinstall raspbian" situation and now I'm afraid click anything in the web interface anymore. 
Thanks!

billz, 2020-05-04:
Hmm, the individual config file paths are all declared in `config.php`, so you could write a little bash script to take a snapshot of them. There's no history of changes made via the UI. 

https://github.com/billz/raspap-webgui/blob/2a7bb37167248d561617850796de8b3f56efdf6a/config/config.php#L10

Alternatively, running the installer again will restore the default configuration. 

Before you reinstall the OS, try enabling logging for hostapd and dnsmasq. This can give you clues if these services fail to work as expected.

Maybe it's an idea to have a 'save state' feature 🤔

frranck, 2020-05-04:
Thanks for the input.
I could run the script again on my first fail, but the second time it was returning without any output. 
I tried the logs from hostapd but they are quite confusing. If you save the state on each apply, there's probably something to do with "patch" to build an history.

billz, 2020-05-05:
I was thinking more along the lines of a one-time snapshot. Not convinced a full state history is needed, unless someone wants to implement it

frranck, 2020-07-04:
Any update on that one?

billz, 2020-09-10:
Bumping this as I think it's a good addition to the project. Just need to carve out some time for it.

closed 2021-01-22 by stale[bot]

raqua, 2022-07-07:
Came here to search for this functionality.