RaspAP/raspap-webgui #577: Changing RASPI_WIFI_CLIENT_INTERFACE constant breaks dashboard

Nikotine1, 2020-05-07:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Instead of eth0, I use a wifi dongle (wlx00c0ca66080c) as wifi client and wlan0 as AP. As per [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#interfaces), I have changed one line in config.php. However, this breaks the dashboard.

**Your environment**
* Raspberry Pi hardware: Pi 3 Model B  
* Raspbian version: Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? both
* Other software or services running with RaspAP? no

**Steps to reproduce**
Setup an AP via wlan0.
Use external wifi dongle to connect to network, don't have ethernet connected.
Follow [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#interfaces) and change config.php.
From:
```
define('RASPI_WIFI_CLIENT_INTERFACE', 'wlan0');
```
To:
```
define('RASPI_WIFI_CLIENT_INTERFACE', 'wlx00c0ca66080c');
```

**Expected behavior**
On the dashboard you see ```wlan0 up```, the traffic via ```wlan0```, the wireless client connection for ```wlx00c0ca66080c``` and connected clients to ```wlan0```.

**Actual behavior**
The dashboard shows ```wlx00c0ca66080c up```, no connected clients and a flat traffic curve.
What does work is the wireless client box, showing the connection of ```wlx00c0ca66080c``` to the wifi router.

Traffic however is recorded for ```wlan0```, as it can be seen on the data usage page.
Also the dhcp page, client list tab is showing correctly the connected clients.

**Screenshots**
N/A

**Additional context**
N/A

billz, 2020-05-08:
Your issue indicates AP and managed mode. In this config the virtual interface `uap0` is displayed in the status indicator, not the client interface as shown. It looks like you've configured a wireless client, but no AP. Please clarify.

Nikotine1, 2020-05-08:
Hi @billz ,
Here are some screenshots, I hope this answers your questions.
If not, please let me know.

[image]

[image]

[image]

billz, 2020-05-08:
You are not in AP-STA mode. Refer to the readme on how to use this feature. 
Review our [issue policy](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#issue-policy) and this statement in our FAQ:

> For these reasons, issues related to wlan0 and wlan1 configurations with external adapters will not be considered.

closed 2020-05-08 by billz

Nikotine1, 2020-05-08:
No I am not, for the simple reason I want to use two different wifi devices. There's no issue with that, the access point works as intended.
I would assume it's a simple code change to show traffic and clients for the AP interface, and client details for the interface that is... the client.
But I understand you want to support this.