RaspAP/raspap-webgui #579: No Internet with NAT mode, can't get IP in bridged mode

jrcichra, 2020-05-09:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [ x] I checked the FAQ before creating this issue
* [ x] I have read and understand the issue reporting guidelines 

**Describe the bug**
On a fresh install of raspap-webgui, connecting with my phone to the wifi network I made doesn't work in either situation (bridged or NAT). It gets an IP with NAT, but says "no internet" and I can't get on the internet. With bridged, my phone says "Obtaining IP Address" and never gets one. Eth0 is plugged in and has a DHCP server listening on the network elsewhere.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  Pi4
* Raspbian version (examples: Buster Lite, Buster Desktop)  Buster Lite(armv7l Linux 4.19.97-v7l+)
* Followed the project prerequisites? (Y/N) Y
* Checked the project FAQ? (Y/N) Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y/N) Y
* Simultaneous AP and managed mode? (Y/N) Y
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**

Go through the quick setup. Only skip OpenVPN and adblocking. Set a hotspot name, password, and put it on channel 6. Save and restart hostapd with the gui buttons. For the bridge bug, just change the slider for bridged, save, and restart.

**Expected behavior**
Either situation would "just work"

**Actual behavior**
"Doesn't work. I either don't get internet or I don't get an address from my upstream DHCP.

billz, 2020-05-09:
> Provide useful detail to reproduce your issue. "Doesn't work" or "not working" is not a valid report.

https://github.com/billz/raspap-webgui/wiki/Reporting-issues

Bridged and NAT modes were [thoroughly tested](https://github.com/billz/raspap-webgui/pull/523#issuecomment-598802431) on the Pi 4 recently. If you have useful debug info such as log entries that would be helpful.

jrcichra, 2020-05-09:
`journalctl -xe` after an attempt to connect (NAT):

```
May 09 10:57:45 justinpi hostapd[2354]: wlan0: STA 80:58:f8:d0:9e:77 IEEE 802.11: associated
May 09 10:57:45 justinpi hostapd[2354]: wlan0: STA 80:58:f8:d0:9e:77 RADIUS: starting accounting session EE78550DC6F19B05
May 09 10:57:45 justinpi hostapd[2354]: wlan0: STA 80:58:f8:d0:9e:77 WPA: pairwise key handshake completed (RSN)
May  9 10:57:45 justinpi dnsmasq-dhcp[2376]: DHCPDISCOVER(wlan0) 80:58:f8:d0:9e:77 
May  9 10:57:45 justinpi dnsmasq-dhcp[2376]: DHCPOFFER(wlan0) 10.3.141.113 80:58:f8:d0:9e:77 
May  9 10:57:45 justinpi dnsmasq-dhcp[2376]: DHCPREQUEST(wlan0) 10.3.141.113 80:58:f8:d0:9e:77 
May  9 10:57:45 justinpi dnsmasq-dhcp[2376]: DHCPACK(wlan0) 10.3.141.113 80:58:f8:d0:9e:77 
May  9 10:57:50 justinpi dhcpcd[2359]: eth0: fe80::9e65:f9ff:fe2a:e790 is reachable again
May  9 10:57:50 justinpi dhcpcd[2359]: eth0: fe80::9e65:f9ff:fe2a:e790 is reachable again
```

I am getting an IP address...which makes me think some forwarding rules are not being applied...

```
sudo iptables -S
-P INPUT ACCEPT
-P FORWARD DROP
-P OUTPUT ACCEPT
-N DOCKER
-N DOCKER-ISOLATION-STAGE-1
-N DOCKER-USER
-N DOCKER-ISOLATION-STAGE-2
-A FORWARD -j DOCKER-USER
-A FORWARD -j DOCKER-ISOLATION-STAGE-1
-A FORWARD -o docker0 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -o docker0 -j DOCKER
-A FORWARD -i docker0 ! -o docker0 -j ACCEPT
-A FORWARD -i docker0 -o docker0 -j ACCEPT
-A FORWARD -o br-54d71ab5ca6d -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -o br-54d71ab5ca6d -j DOCKER
-A FORWARD -i br-54d71ab5ca6d ! -o br-54d71ab5ca6d -j ACCEPT
-A FORWARD -i br-54d71ab5ca6d -o br-54d71ab5ca6d -j ACCEPT
-A DOCKER-ISOLATION-STAGE-1 -i docker0 ! -o docker0 -j DOCKER-ISOLATION-STAGE-2
-A DOCKER-ISOLATION-STAGE-1 -j RETURN
-A DOCKER-USER -j RETURN
-A DOCKER-ISOLATION-STAGE-2 -o docker0 -j DROP
-A DOCKER-ISOLATION-STAGE-2 -j RETURN
```

What other logs would be helpful?

jrcichra, 2020-05-09:
Going to try removing docker, its `iptables` rules might be causing issues...

jrcichra, 2020-05-09:
Yep, the issue was with docker. I had to do `apt purge docker iptables`, reboot to clear the iptables cache, `apt install iptables iptables-persistent`, then finally reinstall raspap-webgui to be safe.

Here are my new iptable rules:

```sudo iptables -S
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
```

This fixed both bridged and NAT modes. You can mark the issue as closed if we don't want to investigate the root cause further :-)

billz, 2020-05-10:
Ah, docker would indeed create an issue. I've been focused on supporting Debian-based distros and haven't had time to look at docker. Would be great to have an option for it. Glad you found a non-dockerized solution in the meantime ;)

Budgiebrain994, 2020-05-15:
Thanks for your consideration on this one - I'd rather not have to remove Docker to run RaspAP.

billz, 2020-05-15:
See #458

jrcichra, 2020-06-12:
Closing with the workaround discussed here: https://github.com/billz/raspap-webgui/issues/458#issuecomment-643420212

Install RaspAP, then install Docker to use them at the same time.

closed 2020-06-12 by jrcichra