RaspAP/raspap-webgui #580: Switching the interface of the access point not working properly

zbchristian, 2020-05-09:
Hi billz,
since I am running with two different wifi interfaces, I need the AP to use wlan1 and wlan0 is the client. There are multiple problems in includes/hostad.php, which prevent the change of the AP interface. 
Main problem is, that RASPI_WIFI_CLIENT_INTERFACE is hardcoded nearly everywhere. I went through and replaced all occurences by $_POST['interface'].

In addition I did changes to adapt the static IP of the chosen interface to match the dhcp range. And the new settings are written back to the corresponding .ini file.

I did not touch any of the settings for briged mode or WifiAP mode. I am not using them and can not test it.

The dhcpcd.conf default value are now read from the "defaults" file instead of the hardcoded values.

In order to get the attached clients listed on the dashboard for an interface other than wlan0, the AP interface has to be extracted. I added this to templates/dashboard.php.

Hope I made all changes clear enough. See attached patch for the details.

Best
  Christian
[0001-Fix-hostapd.php-to-allow-for-a-different-interface-t.patch.zip](https://github.com/billz/raspap-webgui/files/4603946/0001-Fix-hostapd.php-to-allow-for-a-different-interface-t.patch.zip)

billz, 2020-05-10:
Sounds good Christian and thanks for the patch. Is this related to #578?
Will make some time and look at it shortly.

zbchristian, 2020-05-10:
Its actually pretty independent from the general problem with /etc/dhcpdc.conf, except, that the patch already includes the fix for ajax/networking/gen_int_config.php to avoid the wrong order of the options inside of the file. In addition the defaults are read from the "defaults" file to create dhcpcd.conf.

billz, 2020-05-12:
@zbchristian created a PR from your patch.

zbchristian, 2020-05-14:
@billz 
Seems that other people had problems to get RaspAP running with 2 Wifi adapters, so once this patch is applied, it might make sense to add a few words to the documentation.
Here the steps to get this going:
- bind the 2 adapters to the interface names wlan0 and wlan1 
- disable the dynamic assignment: ln -sf /dev/null /etc/systemd/network/99-default.link
- determine the mac addresses of the USB adapters (ifconfig)
- create a udev rules file e.g. /etc/udev/rules.d/72-static-wlanx.rules 
SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="b3:ed:23:23:77:dd", NAME="wlan0"
SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="aa:22:33:55:bb:cc", NAME="wlan1"
- This will guarantee, that the AP works, even if wlan0 is not plugged in. I even replace the client sometimes with a mobile data modem (ppp0). To get this working is a different story.

RaspAP settings:
- set the Hotspot to use wlan1 as the AP
- use wlan0 always as the wifi client (due to the hard coded wlan0 in the code)
- Do not enable the WifiAP mode. This is only needed, when the client and AP are utilizing the same wlan adapter

The patch should take care to assign the static IP to wlan1.
If a dns resolution problem shows up, the static IP of the AP should be added as the dns server of the hotspot and dnsmasq should do the rest.

billz, 2020-05-15:
@zbchristian use of two wifi adapters has been unsupported mainly due to the large number of poor quality dongles on the market. This is discussed on the wiki.

> Important: Be aware that external WiFi adapters (ie, USB "dongles") vary greatly in terms of hardware capabilities and driver support. Many do not work out of the box on the RPi, require a powered USB hub, manual driver and/or firmware installation or are otherwise not well suited for the RPi. For these reasons, issues related to wlan0 and wlan1 configurations with external adapters will not be considered.

Users would often attempt to configure a cheap dongle and raise issues with this project when it fails to work. We can include configuration steps, but with the proviso that this is totally unsupported.

billz, 2020-05-15:
Give the PR a look. I did a quick test pass and found an error in writing `/etc/dhcpcd.conf` that prevents the AP from starting. Have not had time to look further

zbchristian, 2020-05-15:
Concerning the PR:
The problem might be, that I switched to using the /etc/raspap/networking/defaults instead of the hardcode dhcpcd options. If the defaults file does not contain reasonable options to begin with, this will not work. The defaults file is created during installation, but I do not know how. 
If this is the reason, it might the best to go back to the hardcoded values, but this causes again an inconsistence with ajax/networking/get_int_config.php, which is using the defaults file.
My defaults file looks like this:
`# default RaspAP config
hostname
clientid
persistent
option rapid_commit
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
option ntp_servers
require dhcp_server_identifier
slaac private
nohook lookup-hostname
`
This is an exact copy of the options used in the original version of hostapd.php.

zbchristian, 2020-05-15:
Concerning wifi dongles: yes the driver is a real problem, especially for Realtek based devices

zbchristian, 2020-06-04:
Hi Billz,
I checked now the branch zbchristian-wlan-switch and modified it back to the original dhcpcd.conf options. I tested this version with a vanilla raspian lite installation. 
I create a PR for this.

Christian

billz, 2020-06-04:
Merged, thanks

lehni, 2020-06-06:
I am interested in getting this working also, and happy to help out! Out of curiosity: What would it take to have the AP on wlan0 and the Wifi on wlan1? I would like the AP to be on the internal Wifi because that seems to be the one that's the most guaranteed to be up and running.

zbchristian, 2020-06-06:
@lehni The code exists already and is under review by billz (branch zbchristian-wlan-switch). The problem with having wlan0 as the AP and wlan1 as client is, that the wifi client is pretty much hard coded to wlan0. Most of the time as RASPI_WIFI_CLIENT_INTERFACE, which is set in config.php to wlan0, but wlan0 is explicitly used in the code as well. You can try to change it and see, if it works. 
Other option: assign wlan1 to the internal wifi interface via an udev  (see https://github.com/billz/raspap-webgui/issues/580#issuecomment-628506689) and the usb interface to wlan0.

billz, 2020-06-07:
I've committed changes to the feature branch that enable arbitrary switching of the AP interface via the UI. See notes in the PR, thanks

billz, 2020-06-08:
Testers needed for #582. You can (re)install RaspAP with this feature like so:

```
curl -sL https://install.raspap.com | bash -s -- --repo billz/raspap-webgui --branch zbchristian-wlan-switch
```
...or pull the feature branch into an existing installation:

```
cd /var/www/html
sudo git fetch origin
sudo git checkout -b zbchristian-wlan-switch origin/zbchristian-wlan-switch
```

Kudos to @zbchristian for pushing this forward!

zbchristian, 2020-06-09:
@billz I checked the version. The switching of the AP wifi interface works, but the assignment of the wifi client fails, when two separate wifi interfaces are used. 
Reason is, that the client interface `$_SESSION['client_iface']` is set in ` getWifiInterface()` to the wifi interface assigned to the AP (from the hostapd settings). Just forcing `$_SESSION['client_iface']` to the second interface in ` getWifiInterface()` resolves this for now. 
In case, that two wifi interfaces are present, automatically the non-AP one should be used as the client. Will create a PR this.

billz, 2020-06-09:
@zbchristian excellent, I have not tested this in wifi client mode yet. I wanted to verify that switching between wlan0/wlan1 didn't break anything as the session var replaces the constant throughout the project. 

Will check your PR shortly, thanks

billz, 2020-06-09:
PR tested and merged. 

Starting from a clean install on an RPi 4 configured in managed mode, RaspAP is using the onboard wireless `wlan0` interface while hosting an AP with an external adapter on `wlan1`, purely from the UI ie., no manual config changes.

feh123, 2020-06-10:
Hi - yes I would test. I have tried already using the curl command you circulated. I have a +3B using 32 OS and I said y to all including adblock and openvpn. Nothing else is running. My second wifi is a Linksys USB using a RTL8812au chipset. I have a process to compile this and I have had it running successfully in other software as the second RPI wifi.
I have never got to a successful endpoint with an operational hotspot with the new procedure. It seems it does not connect to both wifi's. I do use a wpa_supplicant file to get the client wifi connected from the very start instead of eth0.
Should I try again now with the latest version? Thanks - it would be great to have this running!

feh123, 2020-06-11:
Hi I tried this several times using a Linksys usb dongle (RTL8812au chipset) and a +3B running a fresh install of OS 32 Lite. I said y to all installation questions. I compiled the wifi usb using https://raspberrypi.stackexchange.com/questions/64502/install-drivers-for-rtl8812au-wireless-usb-adapter answer 1.I have had the wifi dongle working in Raspap using the up2smoke procedure. I add a wpa-supplicant to the SD card so the RPI boots and connects to my wifi. Nothing else is installed. 

I have not made it work - on opening the UI I see hotspot inactive. The AP mode icon is ghosted. Wifi Client is correct. But if I try to get the hotspot working by restarting or rebooting then typically the RPI freezes and I lose it from the network. If I enable logging this is the usual output:  

Configurationfile: /etc/hostapd/hostapd.conf

nl80211:Could not configure driver mode

nl80211:deinit ifname=wlan0 disabled_11b_rates=0

nl80211driver initialization failed.

wlan0:interface state UNINITIALIZED->DISABLED

wlan0:AP-DISABLED 

wlan0:CTRL-EVENT-TERMINATING 

hostapd_free_hapd_data:Interface wlan0 wasn't started

Is there anything else I can log to give you more info?

billz, 2020-06-11:
Ran through another clean install on RPi OS (32-bit) Lite with an Edimax EW-7811Un. No special drivers.

1. Configured with wpa_supplicant on boot. No ethernet connection
2. Prerequisites & install feature branch, as above
3. Access UI (no reboot), confirm good wireless client 
4. Configure hostapd interface on wlan1
5. Save settings > Start hostapd
6. Connect several clients (Android + iOS)
7. Speed tests nominal
8. Confirm connected devices + traffic stats on dashboard

Followed up with a baseline compatibility test: connect ethernet, reboot, reconfigure AP on wlan0, restart hostapd, connect several clients. No issues, all steps performed via UI.

feh123, 2020-06-13:
Hi I followed the procedure. Here is my iwconfig before I do (3).
pi@Hotspot2:~ $ iwconfig                                                                              
eth0      no wireless extensions.                                                                     
wlan1     IEEE 802.11  ESSID:"ORBI14"                                                                 
          Mode:Managed  Frequency:5.18 GHz  Access Point: A0:40:A0:6D:87:FE                           
          Bit Rate=200 Mb/s   Tx-Power=31 dBm                                                         
          Retry short limit:7   RTS thr:off   Fragment thr:off                                        
          Power Management:on                                                                         
          Link Quality=70/70  Signal level=-16 dBm                                                    
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0                                    
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0                                    
                                                                                                      
lo        no wireless extensions.                                                                     
                                                                                                      
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"                                                     
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated                                
          Sensitivity:0/0                                                                             
          Retry:off   RTS thr:off   Fragment thr:off                                                  
          Power Management:off                                                                        
          Link Quality:0  Signal level:0  Noise level:0                                               
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0                                    
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

I follow the rest of the steps and go to the UI - see hotspot inactive wlan0 down, hostapd down. I connect wlan0 to my network and wlan0 is up. Then I try to connect wlan1 on hotspot page. No channel # appears and if I change the SSID/password and hit save settings it just reverts to defaults. If I go back to dashboard wlan0 is down. If I look at the network page wlan1 is showing the UI IP address and wlan0 no carrier. 
At some point my changes to the UI freezes the RPI and I need to reboot to restore the UI.
Apologies for a weekend email!

zbchristian, 2020-06-14:
@billz Installed the feature branch from scratch with two wifi interfaces (Edimax 7811Un and Ralink RT5370). Both are supported out of the box by Raspian lite.

Two issues: displayed client interface is not correct on the dashboard and the `$_SESSION` variables `ap_interface` and `wifi_client_interface` are not correctly set, if a dongle is plugged in at a later stage.
Both problem are easily solved. I will send a pull request. 

After the fix everything worked out of the box. The AP interface can now be wlan0 or wlan1 and the the other wifi interface is used as the client.

billz, 2020-06-14:
>  $_SESSION variables ap_interface and wifi_client_interface are not correctly set, if a dongle is plugged in at a later stage

@zbchristian great, I had not tested this case. thanks for catching it.

> After the fix everything worked out of the box. The AP interface can now be wlan0 or wlan1 and the the other wifi interface is used as the client.

Excellent, I'll check out the PR. Think we're getting close to releasing this.

closed 2020-06-16 by billz

feh123, 2020-06-17:
Just to confrim @zbchristian my Linksys usb dongle although apparently ready to use after compiling is not being accepted. I have finally obtained a RT5370 usb wifi and that works well. Thanks for getting such a nice addition working.

zbchristian, 2020-06-17:
@feh123 Great, that it works for you now.
My setup works with usb dongles, for which I had to compile the driver. One is a Edimax EW-7811UTC, which is supported by the rtl8812au (https://github.com/aircrack-ng/rtl8812au) driver and the other a noname AC 1200 dongle support by the rtl88x2bu driver (https://github.com/cilynx/rtl88x2bu). Both work fine with 2.4 and 5Ghz.

feh123, 2020-06-17:
@zbchristian thanks for the info - I may try to use my rtl8812au dongle again as it is faster than the dongle I have installed. But at the moment just nice to have it working.

billz, 2020-06-17:
@feh123 many thanks for your testing efforts.
@zbchristian this is a big enhancement for the project. thanks for pushing it forward.

dixon1e, 2020-06-20:
FWIW I have also installed and tested Edimax EW-7811Un Realtek (8188CUS). Everything worked on a fresh Raspbian Buster install. I ordered several other USB dongles and will try them out when I have a chance. 

Thank you very much @billz @zbchristian @feh123

billz, 2020-06-20:
@dixon1e glad to hear it and appreciate the feedback. Please do share your findings with other adapters. Perhaps we can create a resource for this.

dixon1e, 2020-06-20:
@billz sounds like a great idea. Tested one other just to make sure I wasn't making mistakes.

Tested good: Ralink MT7601U (Brand name ZTC-WUE001-BK)

feh123, 2020-07-01:
I have a Linksys WUSB6300 (RTL8812au) working too thanks to @zbchristian link to aircrack-ng.
Is this branch now merged into the master?