RaspAP/raspap-webgui #585: PiZero clean install: AP but not Wifi client

brettbeeson, 2020-05-14:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
After a fresh install on PiZero Raspian Lite using the quick installer, I can see and connect to the AP and web-interface, but cannot connect to as a client to an AP despite wpa_supplicant.conf being correct.

**Your environment** and **Steps to reproduce**
- fresh install raspbian lite on a RiZero W
- use raspi_config to setup Wifi access, etc. 
- test internet is available via WiFi, and check /etc/wpa_supplicant/wpa_supplicant.conf
- run quick install for raspap. Use default except: No to OpenVPN) 
- Checked [FAQs](https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap) but there is only one adaptor on PiZero so make no changes.
- reboot: looks good (my ip is 10.3.141.1, no boot errors)

- ip addr shows two interfaces: loopback and wlan0 with 10.3.141.1 
- clients can connect to raspAP but cannot ping 10.3.141.1 (!). Running `/etc/raspap/hostapd/servicestart.sh` fixes this. Can now connect a client to RaspAP interface.
- From the instructions I understand I need to connect as a WiFi client first, before I can select Hotspot>Advanced>WiFi client AP mode
- Unit is not connected to internet, although it was before.
- So, using web interface, I try to connect to an existing AP as a client. I can see the wpa_supplicant entry, but "Connect" does not work ("New Network Selected", but no connection)
- Run the `wpa_cli` command (which the interface was running) manually and it fails with "ctril_ifname: wlan0: error No such file or directory"
- Check logs, etc, no error apparent

**Other things I have tried**
- Check FAQs 
- manual install of raspap 
- some success with [the famous blog](https://blog.thewalr.us/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/) but cannot get raspap to work together with this.

**Expected behavior**
Expected to be able to connect as a WiFi client via the web gui, or to retain existing connection via wpa_supplicant.

**Notes**
There are many questions about this issue, and RaspAP has been updated to try to address them. Do we know expect PiZeroW with a fresh install to 'just work'? It's not clear to me if I need to further configure RaspAP (e.g. to deal with have no eth0).
Great project! Thanks!

billz, 2020-05-14:
Per the [readme](https://github.com/billz/raspap-webgui#simultaneous-ap-and-wifi-client): 
> For a device operating in managed mode without an eth0 connection, this configuration must be enabled before a reboot.

brettbeeson, 2020-05-14:
I read that but didn't understand it (my bad). I'd respectfully say I find the docs are confusing. The terminology "managed mode" is multiply defined and [the link to this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#how-do-i-prepare-the-sd-card-to-connect-to-wifi-in-headless-mode) at the end of quick installer appears incorrect.

Its probably obvious, but can you clarify the recommended procedure (PiZeroW, no eth0)? My reading of the docs suggests:
- setup a connection to the internet as a WiFi client on the pi
- follow the docs for prerequisites and [quick install](https://github.com/billz/raspap-webgui#quick-installer), including a "reboot at the end of installation. (maybe don't reboot? is this irretrievably bad?)
- use the web interface to try to connect to WiFi as client (my point of failure)
- select client-WiFi-AP-mode

Thanks for your help - I appreciate it.

billz, 2020-05-19:
The procedure for AP-STA mode on the Pi Zero W is not well documented, that's true. I'll carve out some time and add this to the wiki

MomoSho, 2020-05-20:
> Its probably obvious, but can you clarify the recommended procedure (PiZeroW, no eth0)? My reading of the docs suggests:
>
> * setup a connection to the internet as a WiFi client on the pi
> * follow the [...]

I'm not sure if this will work on the PiZero but for my Pi3, after installing raspap, don't reboot yet, go to the web interface using your Pi's current ip, enable client WiFi AP mode, then reboot. Here's my post with more details documenting what I did when I had the same issue #335 
https://github.com/billz/raspap-webgui/issues/335#issuecomment-630613801

Hopefully other people's comments will also be helpful to you.

brettbeeson, 2020-05-21:
Thanks @billz. I'll give it a spin on a fresh install when you've documented it. 

Thanks for the input @MomoSho: your final setup (i.e. using same chipset for AP and client) sounds pretty similar. 

However, in the meantime, I've got a hybrid-of-two-install-methods working, as [documented here](http://brettbeeson.com.au/pizerow-ap-wifi-client/). But I'm keen to get an 'out-of-the-box' RaspAp going.

billz, 2020-05-23:
Added to the wiki. Will also create a new FAQ item for this
https://github.com/billz/raspap-webgui/wiki/RPi-Zero-W-AP-STA-mode

brettbeeson, 2020-05-24:
Thanks for the detailed instructions. I installed on a fresh PiZeroW without any problems. **It indeed works out of the box**.

It appears _not_ rebooting was the key.  To anyone else with problems on a PiZeroW, try attaching a screen and keyboard. Then you can check everything is working before a reboot:
- `/etc/raspap/hostapd.ini` exists 
- `ip addr` makes sense: mine has two interfaces wlan0 and uap0

Closing issue. Thanks for your time and assistance.

closed 2020-05-24 by brettbeeson

billz, 2020-05-24:
@brettbeeson excellent, glad to hear it.