RaspAP/raspap-webgui #588: Cannot connect to webgui (10.3.141.1)

asnchzv, 2020-05-22:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Cannot connect to webgui (10.3.141.1) to configure AP, but have internet connection.

**Your environment**
* Asus Tinkerboard S
* TinkerOS Debian Stretch (v2.0.11 Kernel: 4.4.132)
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? Onboard 
* Other software or services running with RaspAP? No

**Steps to reproduce**
Quick Install RaspAp on Tinkerboard S with Sixfab RPI 4G/LTE Hat (ppp0 interface)

**Expected behavior**
It supposed to connect to webgui

**Actual behavior**
I have internet connection, can connect over ssh with 10.3.141.1 but can't open webgui.

asnchzv, 2020-05-23:
Solved:

Reinstalling package:
`apt install --reinstall php7.0-cgi`

Then restart service:
`systemctl restart lighttpd`

> ● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-05-23 00:04:02 -04; 1s ago
  Process: 11662 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=0/SUCCESS)
 Main PID: 11686 (lighttpd)
   CGroup: /system.slice/lighttpd.service
           ├─11686 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
           ├─11694 /usr/bin/php-cgi
           ├─11695 /usr/bin/php-cgi
           ├─11696 /usr/bin/php-cgi
           ├─11697 /usr/bin/php-cgi
           └─11698 /usr/bin/php-cgi

billz, 2020-05-23:
Thanks for posting your solution. This is the first install on a Tinkerboard that I'm aware of :)
I suspect the quick installer didn't recognize this flavor of Debian and failed to set php_package option. What is the output of `lsb_release -a` on your TinkerOS?

asnchzv, 2020-05-23:
Hi billz, here's the output:

`No LSB modules are available.`
`Distributor ID:	Debian`
`Description:	Debian GNU/Linux 9.5 (stretch)`
`Release:	9.5`
`Codename:	stretch`

billz, 2020-05-23:
Sure enough, the installer tested for a literal $RELEASE value of "9". I've updated it to be more flexible with minor revisions. Thanks for reporting.

faridbouchdak, 2020-05-24:
Raspberry Pi 3B+
Raspbian Buster Light (releasedate 2020-02-13)
Followed the project prerequisites? Y
Checked the project FAQ? Y
RaspAP Quick Install or Manual setup? Quick Install
Using default configuration? Y
Simultaneous AP and managed mode? Y
Onboard wireless chipset or external adapter? Onboard
Other software or services running with RaspAP? No

Steps to reproduce
Quick Install RaspAp on Raspberri Pi 3B+

When i want to Connect to webgui (http://raspberrypi.local) it returns “Not Authorised”.

faridbouchdak, 2020-05-24:
“Not Authorized” is solved.
Seems to be a Firefox Focus related issue. Safari shows login panel.

billz, 2020-05-24:
@faridbouchdak sounded browser related. good to hear you found a workaround

closed 2020-05-25 by asnchzv