RaspAP/raspap-webgui #590: UI rejects entry of valid passwords for Wifi Clients

ericraymond, 2020-05-24:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I entered a password of 55 characters and it was rejected?  Perhaps the quotes, spaces or other characters are expanded before processing?

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  3B+
* Raspbian version (examples: Buster Lite, Buster Desktop)Buster Lite
* Followed the project prerequisites? (Y/N) Y 
* Checked the project FAQ? (Y/N) Y
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? (Y/N) Y 
* Simultaneous AP and managed mode? (Y/N) Y
* Onboard wireless chipset or external adapter?  Onboard wireless chipset
* Other software or services running with RaspAP?virtualhere

**Steps to reproduce**
Enter a Wifi password that is long (55 chars), has two single quotes (`),  spaces, and perhaps and exclamation mark (!)

**Expected behavior**
The password is saved and optionally used

**Actual behavior**
An error message saying the password must be between 4 and 63 characters?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

ericraymond, 2020-05-24:
Just tried to reproduce this and could not.  Could be state dependent? 

I also reinstalled raspian and raspap-webgui so that may have pull different versions?

Please close.

billz, 2020-05-25:
Unable to repro this. Closing

closed 2020-05-25 by billz

ericraymond, 2020-05-27:
I think this may have been related to the issue "Fix wifi client connect button. Resolves #572"