RaspAP/raspap-webgui #596: Wifi Country Code = Wifi scan broken ?

brunotvaz, 2020-05-28:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
The "wireless scan" doesn't work at all. Had to change file /etc/wpa_supplicant/wpa_supplicant.conf manually

**Your environment**
* Raspberry Pi hardware: Raspberry Pi 4  
* Raspbian version: Raspbian buster desktop (release 10)
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick (without OpenVPN)
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? Both. On-board: Wlan0, External-USB: Wlan1 (Qualcomm Atheros Communications AR9271)
* Other software or services running with RaspAP? None. Default Raspbian install with Raspap only.

**Steps to reproduce**
- Install Raspbian desktop and configure "default settings" like password etc.
- Install Raspap using "curl -sL https://install.raspap.com | bash":
       * Enable all features apart from OpenVPN
- Reboot

**Expected behavior**
I expected to see a full list of SSID's but instead the list was empty

**Actual behavior**
There were no wireless networks shown. Couldn't connect to any network

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I do know that it's not truly supported to have two wireless networks at this stage but I've used previous versions successfully in the past.

brunotvaz, 2020-05-28:
(additional findings)

Since I couldn't connect to any wireless network because the "scan" wasn't showing anything I decided to launch the desktop and had a look.
There I clicked the network icon and tried to configure a wireless network.
It was asking me to configure the country code (which I had before).
After choosing GB and saving the setting nothing changed. I mean it always asks for country code and won't let you pic scanned networks around you.
[image]

Then I decided to use raspi-config instead and it also failed to set the country code (or any wireless settings for that matter).
[image]
[image]
[image]

[image]
[image]
[image]

The only way I could get my wireless networks working was by changing manually the file:
**/etc/wpa_supplicant/wpa_supplicant.conf**
`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=GB

network={
        ssid="my_private_wireless_1"
        psk="my_private_wireless_1_passcode"
        key_mgmt=WPA-PSK
}

network={
	ssid="my_private_wireless_2"
	psk="my_private_wireless_2_passcode"
}
`

brunotvaz, 2020-05-28:
So... after configuring the wireless networks manually I still didn't had any internet access.
Since I'm not using eth1, I'm guessing the reason is to due with the "hard coded" settings of the RaspAp application.
I mean, my network settings were like:
[image]
[image]
[image]
That would never work for me because eth0 was not plugged in and also my "internet network" was wlan1 instead of the "hard coded" eth0 default gateway" setting.

brunotvaz, 2020-05-28:
To make it work (in my scenario) I had to do the following steps:
[image]
[image]
[image]
[image]

With those settings I've managed to enable internet access via the eth0 network and therefore all devices in that network (192.168.2.xxx) have internet access.

**NOTE: I cannot use the Raspap "save" options because it breaks the above configurations back to the "hard coded settings".**

brunotvaz, 2020-05-28:
Summary:

I have it working at the moment but it's extremely limited. I cannot:
- Scan for wireless networks
- Change "networking" settings using the RaspAp app because it defaults to the "eth0 scenario"
- Change "hotspot" settings using the RaspAp app because it defaults to the "eth0 scenario"
- Change "dhcp server" settings using the RaspAp app because it defaults to the "eth0 scenario"

If I may can I suggest a couple of things?
**- Have the RaspAp installer ask which IP to use for the hotspot instead of the "hard coded" 10.3.141.1 by default.
- Have the RaspAp installer ask which is the "internet gateway interface"  instead of assuming eth0 or prevent hard coded values altogether** 
I have read the following: #https://github.com/billz/raspap-webgui/issues/580 and it seems that findings are somehow correct. There are "hard coded values" that break the app if not using the "default eth0/wlan0 scenario".

I'm sorry for the long post(s). Just trying to help @billz and the community as much as I can.

billz, 2020-05-29:
Refer to [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#wifi-scanning-doesnt-work-or-i-get-the-error-cannot-execute-wpa_cli-reconfigure-help) which addresses this.

The issue with wpa_supplicant is specific to Raspbian; other Debian-based distros do not exhibit this problem. I have not had time to dig into the root cause and would rather not build in a 
workaround for a poor implementation. For now, it's a manual fix as described in the FAQ.

The FAQ also explains why configurations with external adapters are unsupported.

Changing the default config at install time is unlikely to happen. The vast majority of RaspAP's users want to get a working AP up quickly with a 'known good' base config that can then be modified via the UI. This is a guiding principle of the project. Changing the network config at install time invites potential conflicts which leads to more troubleshooting, bug reports, etc. The harsh reality is I don't have time for this.

Anyway, welcome!

billz, 2020-06-09:
To the core issue re: wlan0/wlan1, this should be resolved with #582.

closed 2020-06-09 by billz