RaspAP/raspap-webgui #601: Quick installer doesn't work on lastest Raspberry Pi OS (May 2020)

yeshwantsingh, 2020-06-01:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I tried to install raspap on latest raspberry pi os using quick installer, but it fails pointing dependencies errors along the way.

**Your environment**
* Raspberry Pi hardware: Pi 4
* Raspbian version: Buster Desktop
* Followed the project prerequisites? : Y
* Checked the project FAQ? : Y
* RaspAP Quick Install or Manual setup? : Quick install
* Using default configuration? : Y
* Simultaneous AP and managed mode? : Y
* Onboard wireless chipset or external adapter? : Onboard 
* Other software or services running with RaspAP? : Fresh install

**Steps to reproduce**
Run quick installer on raspberry pi os (may 2020) running on Pi 4.

billz, 2020-06-01:
The installer runs cleanly on Raspberry Pi OS (32-bit) Lite. The 64-bit beta release is unsupported. Which distro are you using?

yeshwantsingh, 2020-06-02:
I'm using Raspberry Pi OS (32-bit) with desktop and recommended software (May 2020 release)

billz, 2020-06-02:
Per the readme, this project recommends the Lite version.

The distro you are using was released only days ago. It is untested with this project and, quite honestly, I won't have time to look at it for several weeks, if I don't forget about it or if something more urgent doesn't take its place.

You can either a) use the Lite version as suggested, or b) contribute to the project by improving it.

closed 2020-06-14 by billz