RaspAP/raspap-webgui #602: Feature: External Api to manage RaspAp Configurations

fjhunter, 2020-06-02:
Hello everybody,

I would like to use RaspAp as Hotspot and as WLAN client for an IoT Network. 
A Server should take care of managing this network. 
So I can`t use the gui to change the RaspAp settings. I need to change them externally. It would be great to get all current settings via REST Api and be able to change them.

## What needs to be done 

* Additional Rest Endpoints which return json instead of rendered html(e.g.: get all available wifi stations).
* A [Swagger documentation](https://swagger.io/) for the REST endpoints. With which you can automatically create your REST client

billz, 2020-06-03:
Sounds great. What can you contribute?

fjhunter, 2020-06-03:
I started with the swagger documentation, when I noticed, that some rest endpoints are missing. I'm planning on adding the additional rest endpoints and finishing the swagger documentation in the next few weeks

closed 2020-08-09 by stale[bot]

alexszilagyi, 2021-03-03:
@fjhunter : Thanks for your contribution! Could you please tell me the endpoint URL? Something like: `https://10.3.141.1/api/` ?