RaspAP/raspap-webgui #605: Empty connected devices box

profwyattb, 2020-06-06:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
When connecting devices to the hotspot, devices do not appear in the connected devices box. However, they appear in DHCP Server -> Client list

**Your environment**
* Rasberry Pi 3B
* Buster Lite
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* Quick Install
* Using default configuration? (N)
* Simultaneous AP and managed mode? (N)
* Onboard wireless chipset for hostspot, dongle for connecting to router
* Stock

**Steps to reproduce**
1. Configure RaspAP to use internal Wifi as hotspot and configure only the dongle to connect to the router

**Expected behavior**
A list of connected devices in the connected device box on the homepage

**Actual behavior**
Empty connected devices box

**Screenshots**
[image]
[image]

billz, 2020-06-06:
DHCP leases and connected devices are two different things.

> configure only the dongle to connect to the router

Are you using an external wifi adapter?

profwyattb, 2020-06-06:
I'm using an external wifi adapter for the connection to the main router and the internal wifi for the hotspot. Hope this helps :)

billz, 2020-06-07:
This widget is populated using `RASPI_WIFI_CLIENT_INTERFACE`. Set this value to your adapter interface  or wait for #582

profwyattb, 2020-06-07:
Just to clarify, the connected devices box meant to be populated with devices that are connected to the Raspberry Pi AP?

So if I have:
wlan0 = internal wifi - used for hosting the AP
wlan1 = wifi dongle - used to connect to the real AP with an internet connection
What should `RASPI_WIFI_CLIENT_INTERFACE` be set to. Currently, it's set to `wlan1`

Thanks :)

billz, 2020-06-08:
> the connected devices box meant to be populated with devices that are connected to the Raspberry Pi AP?

correct. in your case, this is wlan0

closed 2020-07-23 by billz