RaspAP/raspap-webgui #610: AP is not shown on Fire Tablets

svenlimbach, 2020-06-12:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I successfully configured RaspAP on a RPI4. Apple Devices can see the WLAN and connect. Android based Amazon Fire Tablets not see the WLAN at all.

Anything i missed during setup?

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  -> RPI4 4GB
* Raspbian version (examples: Buster Lite, Buster Desktop) -> Buster Lite
* Followed the project prerequisites? (Y/N) -> Y
* Checked the project FAQ? (Y/N) -> Y
* RaspAP Quick Install or Manual setup? -> Quick
* Using default configuration? (Y/N) -> Y
* Simultaneous AP and managed mode? (Y/N) -> Y
* Onboard wireless chipset or external adapter? -> OnBoard
* Other software or services running with RaspAP? -> Only RaspAP

**Steps to reproduce**
Install RaspAP via Quick Install.

**Expected behavior**
All devices should see the AP WLAN

**Actual behavior**
Only Apple Devices can see it.

billz, 2020-06-12:
I'll do your homework for you for a suitable fee.

svenlimbach, 2020-06-12:
Thx for the feedback. I will take a look again what i missed ;-)

closed 2020-07-29 by svenlimbach