RaspAP/raspap-webgui #612: OpenVPN: partial or no internet access

NikaZhenya, 2020-06-12:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**

Hi, I am trying to enable the VPN and, even though the config is working, I can only _sometimes_ access to duckduckgo and Telegram. The rest of websites o email client connections are denied. Most of the times I don't have any internet access.

**Your environment**
* Raspberry Pi 3 Model B Rev 1.2
* Raspbian Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset

**Steps to reproduce**

1. Add OpenVPN conf file (see below)
2. Add user and pass
3. Start OpenVPN

```
$ cat /etc/openvpn/client/client.conf
remote 198.252.153.226
remote vpn.riseup.net
auth SHA256
auth-nocache
auth-user-pass login.conf
cipher AES-256-CBC
client
dev tun
nobind
persist-key
persist-tun
port 80
proto udp
remote-cert-tls server
resolv-retry infinite
script-security 2

<ca>
-----BEGIN CERTIFICATE-----
MIIF2jCCA8KgAwIBAgIIVogyQTSIzc8wDQYJKoZIhvcNAQELBQAwgYYxGDAWBgNV
BAMTD1Jpc2V1cCBOZXR3b3JrczEYMBYGA1UEChMPUmlzZXVwIE5ldHdvcmtzMRAw
DgYDVQQHEwdTZWF0dGxlMQswCQYDVQQIEwJXQTELMAkGA1UEBhMCVVMxJDAiBgkq
hkiG9w0BCQEWFWNvbGxlY3RpdmVAcmlzZXVwLm5ldDAiGA8yMDE2MDEwMjIwMjU0
MFoYDzIwMjYwMzMwMjAyNjAxWjCBhjEYMBYGA1UEAxMPUmlzZXVwIE5ldHdvcmtz
MRgwFgYDVQQKEw9SaXNldXAgTmV0d29ya3MxEDAOBgNVBAcTB1NlYXR0bGUxCzAJ
BgNVBAgTAldBMQswCQYDVQQGEwJVUzEkMCIGCSqGSIb3DQEJARYVY29sbGVjdGl2
ZUByaXNldXAubmV0MIICIjANBgkqhkiG9w0BAQEFAAOCAg8AMIICCgKCAgEAw2VV
uoz4xqeB1ROIwXBRaj0prOqEFX89A7+2rslGRfjM8NPHyBLGleoHTK3DPwadtQeg
ulaEOAjM5EMXTEX/o9H46L6h729HUWPCwVssvvOjyxTyGJDf7Ihd/Ab7ODtlJSyc
g31aXMioA5pGz5QnS3VGz4nE9+NL+jobc/NbhaacsEPR/7xO7meRNu/1S+YiHK1y
BSVrfap3XItlcNHDGNQkPyyJbS3pAS1lQs2HCBTzcFCamCkDOC7cRh9wZ4GH8U2f
2s0mDD5zhRpheNW4gFBtGpqHiRXv7WJW612aaXzKQQoIq2loGNvOpnyBPKL3jjUT
Rxv5IzWMV0nAofMCy25u/S4J65uSEd9mLNXFJ3rl+cFaybcOUXktTbS7bZy6cMyf
/gO28bEXIWr5WfZf8jCbPyOVfExZquG3aS+0YPWmIJCheXQzgiwplZy93oND1GGQ
f+1R2F7GPwNXQdefv2xm7PTWhHbSWHHmeY89qYED+yFJrX5ChoFoBbYs1lMmdU/C
2MnQBFtvcVockXFAUONyMKiq8ZP6sQ1lu0rO9Bvkhx55sJLZOmjN3g4S1K97PbbI
5DzHKcR0JQSt8ZtCY/MuMbwvlNYo98bFWvlfKET0KPtogNNH0PNfJmStKR8jWGjE
HnUNXo7YDfK90iEKTjLz2K5CYzH5Dm6iYJNaaykCAwEAAaNGMEQwEgYDVR0TAQH/
BAgwBgEB/wIBADAPBgNVHQ8BAf8EBQMDBwQAMB0GA1UdDgQWBBTGek7ebtq2Ibm+
2K6je1IMobvEkzANBgkqhkiG9w0BAQsFAAOCAgEAO2B3jnL+8LeoRkc282qUpHyu
[... 126 lines cut ...]
Jun 12 15:44:07 raspberrypi openvpn[2377]: Initialization Sequence Completed
Jun 12 15:45:31 raspberrypi openvpn[2377]: event_wait : Interrupted system call (code=4)
Jun 12 15:45:31 raspberrypi openvpn[2377]: /sbin/ip addr del dev tun0 172.27.0.26/22
Jun 12 15:45:31 raspberrypi openvpn[2377]: SIGTERM[hard,] received, process exiting
Jun 12 15:45:41 raspberrypi openvpn[2689]: WARNING: file 'login.conf' is group or others accessible
Jun 12 15:45:41 raspberrypi openvpn[2689]: OpenVPN 2.4.7 arm-unknown-linux-gnueabihf [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] built on Feb 20 2019
Jun 12 15:45:41 raspberrypi openvpn[2689]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Jun 12 15:45:41 raspberrypi openvpn[2689]: TCP/UDP: Preserving recently used remote address: [AF_INET]198.252.153.226:80
Jun 12 15:45:41 raspberrypi openvpn[2689]: UDP link local: (not bound)
Jun 12 15:45:41 raspberrypi openvpn[2689]: UDP link remote: [AF_INET]198.252.153.226:80
Jun 12 15:45:42 raspberrypi openvpn[2689]: [vpn.riseup.net] Peer Connection Initiated with [AF_INET]198.252.153.226:80
Jun 12 15:45:44 raspberrypi openvpn[2689]: TUN/TAP device tun0 opened
Jun 12 15:45:44 raspberrypi openvpn[2689]: /sbin/ip link set dev tun0 up mtu 1500
Jun 12 15:45:44 raspberrypi openvpn[2689]: /sbin/ip addr add dev tun0 172.27.0.32/22 broadcast 172.27.3.255
Jun 12 15:45:44 raspberrypi openvpn[2689]: Initialization Sequence Completed
```

```
$ sudo iptables -S
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
-A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i wlan0 -o tun0 -j ACCEPT
```

billz, 2020-06-13:
The logs indicate that this project has successfully configured an openvpn client for you. Great!

>  even though the config is working, I can only sometimes access [snip]

Intermittent access is an issue for your openvpn provider, _not_ this project. It's likely your provider is oversubscribed. There are various Linux tools such as `mtr` or `traceroute` you can use to measure network latency and make your own assessment.

Protip: you usually get what you pay for with 'no-cost' public VPNs, which is what riseup.net appears to offer.

NikaZhenya, 2020-06-13:
Thanks for your help. I am glad it is not and issue of this great project. Cheers!

closed 2020-06-13 by NikaZhenya