RaspAP/raspap-webgui #613: questions about hotspot and vpn client

ninjajs, 2020-06-13:
Hi, I installed raspap-webgui on a raspberry pi 4 to use as a hotspot with vpn.

1. After I installed raspap-webgui I can no longer connect to any wifi networks from the pi (I have to use the wired ethernet). Is it possible to use wifi instead of the wired cable?

2. After enabling openvpn all traffic seems to be routed through the vpn. Is it possible to only route traffic from devices that are connect to the hotspot?

svenlimbach, 2020-06-13:
> Hi, I installed raspap-webgui on a raspberry pi 4 to use as a hotspot with vpn.
>
> 1. After I installed raspap-webgui I can no longer connect to any wifi networks from the pi (I have to use the wired [...]

Regarding 1:

Did you configure the WiFi Client AP Mode after installation? You can find this in the FAQ

https://github.com/billz/raspap-webgui/wiki/RPi-Zero-W-AP-STA-mode

svenlimbach, 2020-06-13:
> Hi, I installed raspap-webgui on a raspberry pi 4 to use as a hotspot with vpn.
>
> 1. After I installed raspap-webgui I can no longer connect to any wifi networks from the pi (I have to use the wired [...]

Regarding 2

RaspAP acting as OpenVPN Client. The VPN routing is configured in your OpenVPN Client/Server config. Please check your OpenVPN Config for the correct routing you want to be in place, when the VPN is connect

https://unix.stackexchange.com/questions/263678/openvpn-understand-the-routing-table-how-to-route-only-the-traffic-to-a-spec

OpenVPN has a directive for adding and removing of routes client side in your OpenVPN config file with the route option.

Adding:
route 172.16.0.0 255.255.255.0 -> Replace the network with yours behind the OpenVPN Server

to your OpenVPN config file on the VPN client. This will add the route automatically when you connect.

At the moment I would guess is 0.0.0.0/0 the default route for your OpenVPN Config

ninjajs, 2020-06-14:
> Regarding 1:
>
> Did you configure the WiFi Client AP Mode after installation? You can find this in the FAQ
>
> https://github.com/billz/raspap-webgui/wiki/RPi-Zero-W-AP-STA-mode

The WiFi Client AP Mode option is disabled for me.

billz, 2020-06-14:
@ninjajs the readme states clearly that AP-STA mode is enabled _after_ you configure a wireless client. Alternatively, you can wait for the wlan0 + wlan1 feature release or, better still, participate in testing. 

Please read [this](https://github.com/billz/raspap-webgui/wiki/Reporting-issues) in its entirety and understand that this is not a troubleshooting forum. There is a wealth of info available on our wiki and FAQ.

closed 2020-07-11 by billz