RaspAP/raspap-webgui #615: Raspberry connected to router via UAP but sometimes cannot connect

sunberrysolar, 2020-06-14:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Raspberry use wificlient to connect to my router, with UAP mode to keep a direct connection with my phone if router is off. 
My router config page show my raspberry at 192.168.1.188. Sometimes I can connect to it, but sometimes not, even if it is showing.

**Your environment**
* Pi 3 Model B
* Raspbian version (examples: Buster Lite, Buster Desktop)
* Followed the project prerequisites
* Checked the project FAQ
* RaspAP Quick Install
* Using default configuration without DHCP and OpenVPN
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset
* Other software or services running with RaspAP : Jeedom

**Steps to reproduce**

**Expected behavior**
Able to connect everytime

**Actual behavior**
Sometimes showing as connected by router but unreachable

sunberrysolar, 2020-06-14:
[image]
[image]

sunberrysolar, 2020-06-14:
Please does not work on this before I confirm, it seems router fixed IP by  MAC Adress config does not work properly

billz, 2020-06-14:
"Sometimes" is not a valid bug report. RaspAP has performed its function by creating a bridged AP. Thereafter it is up to _you_ to manage your network. 
https://github.com/billz/raspap-webgui/wiki/Bridged-AP-mode

closed 2020-06-14 by billz