RaspAP/raspap-webgui #617: update 2.3.1 to 2.4.1

defaliz, 2020-06-23:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
A clear and concise description of what the bug is.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  
* Raspbian version (examples: Buster Lite, Buster Desktop)
* Followed the project prerequisites? (Y/N)
* Checked the project FAQ? (Y/N)
* RaspAP Quick Install or Manual setup?
* Using default configuration? (Y/N)
* Simultaneous AP and managed mode? (Y/N)
* Onboard wireless chipset or external adapter? 
* Other software or services running with RaspAP?

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
Tell us what you observed instead.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

defaliz, 2020-06-23:
hi, how upgrade v2.3.1 to v2.4.1 ? and keeping my configuration ?
regards

GiannisDorokidis, 2020-06-24:
https://github.com/billz/raspap-webgui/wiki/FAQs#what-are-the-steps-to-upgrade-raspap

defaliz, 2020-06-24:
great !
thank you very much..

billz, 2020-06-24:
Version 2.4 includes ad blocking, which needs some [extra steps to be performed](https://github.com/billz/raspap-webgui/wiki/Manual-installation) that aren't covered by the FAQ.

An `--upgrade` option for the installer would simplify things.

billz, 2020-06-26:
Quick installer wiki page + FAQ are updated. 
tl;dr: run `curl -sL https://install.raspap.com | bash -s -- --upgrade`

@defaliz thanks for reporting

PR #619 (2020-06-26): Adds upgrade option to installer
Resolves #617
commit 199065e9 (2020-06-26): Update installer with upgrade option. Resolves #617
closed 2020-06-26 by billz