RaspAP/raspap-webgui #618: Addition to the Network UI: Information about the current internet connectivity

zbchristian, 2020-06-25:
I had frequently the problem, that I wanted to know the IP address assigned to the raspi, the gateway and the connectivity of the uplink to the internet. Especially, if connectivity exists, but DNS resolution fails.
I added the corresponding information to the Network UI (see image below). the connectivity is tested via ping to the cloudfare DNS server (1.1.1.1 and one.one.one.one). 
If of general interest, I can provide a pull request.

[image]

billz, 2020-06-28:
Nice work, I think others would find this useful. If you create a PR I'll give it a test. Thanks

billz, 2020-06-29:
@zbchristian I've pushed some commits to your feature branch. give them a look, we can merge this when done.

closed 2020-06-29 by billz

tped, 2020-09-06:
Very nice feature (and software, I may add!)

Mostly an FYI in case you are scratching your head on internet connectivity issues in the future ... but ...

Our ISP (TDS) blocks cloudflare 1.1.1.1!  I originally found this when Dietpi starting pinging 1.1.1.1 to test internet connectivity.   - Here's one of the articles: (https://actiontecsupport.zendesk.com/hc/en-us/community/posts/360056087832-T3200-DNS-blackholed-for-Cloudflares-1-1-1-1-and-1-0-0-1).  Actiontec/TDS issue is a looong story/debate, I simply avoid 1.1.1.1.

I changed  /var/www/html/includes/defaults.php to get it working.  Internet access works fine even though Internet Connection screen still shows 1.1.1.1/one.one.one.one with red Xs.  I can dig into that part and submit bug/change form if needed.

changed ...
'RASPI_ACCESS_CHECK_IP' => '1.1.1.1',
'RASPI_ACCESS_CHECK_DNS' => 'one.one.one.one',

to ...
 'RASPI_ACCESS_CHECK_IP' => '9.9.9.9',
  'RASPI_ACCESS_CHECK_DNS' => 'dns.quad9.net',

May help others

frennkie, 2022-04-19:
@billz @zbchristian I really like this addition! :-D  

Would it also be possible to display the full routing table at the bottom? As far as I see it "Current settings" simply show's the standard output for each interface (ignoring the loopback) of `ip address show`. Could the output of `ip route show` be added?

billz, 2022-04-19:
@frennkie an IP routing table is a **great** idea and would be very useful in this context.
I'll carve out some time and work on a prototype. many thanks for the suggestion.

billz, 2022-04-21:
The **Internet connection** table also uses `ip route list` however the output is processed to render a more user-friendly summary. 

Below this, the **Routing table** section simply dumps the same command output but in a raw form. It's displayed with pre-formatted text similar to the interface details. 

[image]

@frennkie @zbchristian thoughts on this?

frennkie, 2022-04-21:
Looks good to me..! :-) 👍