RaspAP/raspap-webgui #622: Security issue wifi qr code

CrashPlayer, 2020-06-29:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
If you open the url 
http://[IP_HERE]/app/img/wifi-qr-code.php
the qr code for login will show up even if you are not logged in.
That means pure IP-based access to the raspberry pi will allow you to log into the AP without credentials.

**Your environment**
* Raspberry Pi hardware: Raspberry Pi 3B
* Raspbian version: Buster
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? Onboard WIFI
* Other software or services running with RaspAP? No

**Steps to reproduce**
-Install RaspAP on your Raspberry Pi
-Go into yur browser and goto url **http://[IP_HERE]/app/img/wifi-qr-code.php**
-scan the shown qr code with your mobile phone

**Expected behavior**
I expected RaspAP-webgui to ask for credentials here.

**Actual behavior**
I saw the login code without authentication.

**Screenshots**
Sorry, screenshots have a problem on my phone

**Additional context**
Please apply authentication requirement on that image file.

billz, 2020-07-10:
Thanks for reporting :)

closed 2020-07-10 by billz

CrashPlayer, 2020-12-06:
Still possible to use if you load the startpage but you dont input credentials so it shows you "Not authorized". Then you edit the HTML of the page's body and you put in
`<img src="/app/img/wifi-qr-code.php" class="figure-img img-fluid" alt="RaspAP Wifi QR code" style="width:100%;">`
That's the original HTML code used in templates/hostapd/security.php to call the QR-Code and because weather you log in r not, the QR-Code PHP doesn't check the credentials, but checks if it's being called directly or through another document. So you could call it from within any document, even from another server. Maybe you could use the QR-Code-PHP-File to generate not the image but the Data-URL for the QR-Code to make it only usable from the PHP-Skripts on the server itself.
I hope that's enough information to fix it.

CrashPlayer, 2020-12-09:
(Posted because #726 was closed as duplicate but is a problem that belongs here)
As in duplicated issue #726 mentioned:
QR-Code accessible by accessing it as resource in another webpage
Paste:
`<img src="http://[IP-HERE]/app/img/wifi-qr-code.php" class="figure-img img-fluid" alt="RaspAP Wifi QR code" style="width:100%;">`
QR-Code will appear

commit 7af8b302 (2020-07-10): Prevent direct file access. Resolves #622