RaspAP/raspap-webgui #627: Manual installation does not enable FastCGI module [solution included]

massimiliano-leoni, 2020-07-04:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
After following the manual installation instructions, the access point was present and working but the web interface was unreachable, giving a "403 - forbidden" error.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  Pi 4
* Raspbian version (examples: Buster Lite, Buster Desktop) Buster Desktop
* Followed the project prerequisites? (Y/N) Y
* Checked the project FAQ? (Y/N) Y
* RaspAP Quick Install or Manual setup? Manual
* Using default configuration? (Y/N) Y
* Simultaneous AP and managed mode? (Y/N) N
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? Nothing to report

**Steps to reproduce**
Follow manual installation steps. Log to the web ui.

**Expected behavior**
See the web ui.

**Actual behavior**
Get 403 - Forbidden

**Solution**
I was able to solve the issue adding the FastCGI module to the lighttpd.conf file, following https://www.cyberciti.biz/tips/lighttpd-php-fastcgi-configuration.html .

billz, 2020-07-04:
Per the docs, this is handled with `sudo lighttpd-enable-mod fastcgi-php`

massimiliano-leoni, 2020-07-04:
Oh dear! I checked my logs and somehow I skipped that block :_D
Never mind then, please disregard this issue, sorry for the false alarm.

closed 2020-07-04 by massimiliano-leoni