RaspAP/raspap-webgui #631: Intermittent performance (loss of connection)

asnchzv, 2020-07-13:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I have intermittent performance with the AP, when it loses connection I cannot enter to webgui.

**Your environment**
* Asus Tinkerboard S
* TinkerOS Debian Stretch (v2.0.11 Kernel: 4.4.132)
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? Onboard 
* Other software or services running with RaspAP? No

**Steps to reproduce**
Quick Install RaspAp on Tinkerboard S with Sixfab RPI 4G/LTE Hat (ppp0 interface)

**Expected behavior**
It supposed to have permanent internet connection

**Actual behavior**
When it loses internet connection also doesn't connect to webgui

Some outputs:

- dhclient:
> root@tinkerboard157:~# dhclient -v
> Internet Systems Consortium DHCP Client 4.3.5
> Copyright 2004-2016 Internet Systems Consortium.
> All rights reserved.
> For info, please visit https://www.isc.org/software/dhcp/
> 
> Listening on LPF/wlan0/a2:92:6e:25:cc:0e
> Sending on   LPF/wlan0/a2:92:6e:25:cc:0e
> Listening on LPF/eth0/04:d9:f5:55:76:17
> Sending on   LPF/eth0/04:d9:f5:55:76:17
> Sending on   Socket/fallback
> DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
> DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
> DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 11
> DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
> DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 12

- dhcpcd:
> root@tinkerboard157:~# systemctl status dhcpcd -l
> ● dhcpcd.service - LSB: IPv4 DHCP client with IPv4LL support
>    Loaded: loaded (/etc/init.d/dhcpcd; generated; vendor preset: enabled)
>    Active: active (running) since Mon 2020-07-13 16:53:54 -04; 37min ago
>      Docs: man:systemd-sysv-generator(8)
>   Process: 2019 ExecStart=/etc/init.d/dhcpcd start (code=exited, status=0/SUCCESS)
>    CGroup: /system.slice/dhcpcd.service
>            └─2158 /sbin/dhcpcd
> 
> Jul 13 17:31:19 tinkerboard157 dhcpcd[2158]: wlan0: deleting address fe80::59c3:e61a:b9d3:b2ee
> Jul 13 17:31:19 tinkerboard157 dhcpcd[2158]: wlan0: carrier acquired
> Jul 13 17:31:19 tinkerboard157 dhcpcd[2158]: wlan0: adding address fe80::59c3:e61a:b9d3:b2ee
> Jul 13 17:31:19 tinkerboard157 dhcpcd[2158]: wlan0: IAID 6e:25:cc:0e
> Jul 13 17:31:19 tinkerboard157 dhcpcd[2158]: wlan0: probing address 10.3.141.1/24
> Jul 13 17:31:20 tinkerboard157 dhcpcd[2158]: wlan0: soliciting an IPv6 router
> Jul 13 17:31:24 tinkerboard157 dhcpcd[2158]: wlan0: using static address 10.3.141.1/24
> Jul 13 17:31:24 tinkerboard157 dhcpcd[2158]: wlan0: adding route to 10.3.141.0/24
> Jul 13 17:31:24 tinkerboard157 dhcpcd[2158]: wlan0: removing route to 10.3.141.0/24
> Jul 13 17:31:33 tinkerboard157 dhcpcd[2158]: wlan0: no IPv6 Routers available

- hostapd:
> root@tinkerboard157:~# systemctl status hostapd -l
> ● hostapd.service - LSB: Advanced IEEE 802.11 management daemon
>    Loaded: loaded (/etc/init.d/hostapd; generated; vendor preset: enabled)
>    Active: active (running) since Mon 2020-07-13 16:53:46 -04; 28min ago
>      Docs: man:systemd-sysv-generator(8)
>   Process: 1881 ExecStart=/etc/init.d/hostapd start (code=exited, status=0/SUCCESS)
>    CGroup: /system.slice/hostapd.service
>            └─1909 /usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf
> 
> Jul 13 17:03:19 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 WPA: pairwise key handshake completed (RSN)
> Jul 13 17:03:45 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 WPA: group key handshake completed (RSN)
> Jul 13 17:12:55 tinkerboard157 hostapd[1909]: wlan0: STA e4:ce:8f:39:91:70 IEEE 802.11: disassociated
> Jul 13 17:13:47 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 IEEE 802.11: disassociated
> Jul 13 17:13:47 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 IEEE 802.11: associated
> Jul 13 17:13:47 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 RADIUS: starting accounting session 5F0CC9DA-00000007
> Jul 13 17:13:47 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 WPA: pairwise key handshake completed (RSN)
> Jul 13 17:20:47 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 IEEE 802.11: associated
> Jul 13 17:20:48 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 RADIUS: starting accounting session 5F0CC9DA-00000008
> Jul 13 17:20:48 tinkerboard157 hostapd[1909]: wlan0: STA 24:0a:c4:c8:09:d0 WPA: pairwise key handshake completed (RSN)

- lighttpd
> root@tinkerboard157:~# systemctl status lighttpd -l
> ● lighttpd.service - Lighttpd Daemon
>    Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
>    Active: active (running) since Mon 2020-07-13 16:53:41 -04; 23min ago
>  Main PID: 806 (lighttpd)
>    CGroup: /system.slice/lighttpd.service
>            ├─806 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
>            ├─920 /usr/bin/php-cgi
>            ├─939 /usr/bin/php-cgi
>            ├─940 /usr/bin/php-cgi
>            ├─941 /usr/bin/php-cgi
>            └─942 /usr/bin/php-cgi
> 
> Jul 13 16:54:09 tinkerboard157 lighttpd[806]: hostname: No address associated with hostname
> Jul 13 16:54:14 tinkerboard157 lighttpd[806]: hostname: No address associated with hostname
> Jul 13 16:54:42 tinkerboard157 lighttpd[806]: hostname: No address associated with hostname

billz, 2020-07-15:
TinkerOS is untested and unsupported.

Check dmesg and syslog for clues as to why wlan0 is dissociating. You can also enable verbose logging in hostapd.conf with `logger_syslog_level=0` and run it interactively.

Alternatively, Armbian is [recommended for RaspAP](https://github.com/billz/raspap-webgui#supported-operating-systems) and also compatible with Tinkerboard.

gstrauss, 2020-07-29:
> Jul 13 16:54:09 tinkerboard157 lighttpd[806]: hostname: No address associated with hostname

It looks like DNS is failing and lighttpd is reporting that it is unable to resolve the hostname to an IP address, and therefore does not know to what IP address lighttpd is to `bind()` and `listen()`.  Change your lighttpd.conf to listen on `*` or on an IP address rather than on a hostname.

asnchzv, 2020-07-30:
> It looks like DNS is failing and lighttpd is reporting that it is unable to resolve the hostname to an IP address, and therefore does not know to what IP address lighttpd is to `bind()` and [...]

Thanks for your reply, but I decided to make a clean install and it works perfectly!

closed 2020-07-30 by asnchzv