RaspAP/raspap-webgui #633: dhcp server won't start because of bad 'log-facility' line

jginspace, 2020-07-15:
* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [x ] I checked the FAQ before creating this issue
* [x ] I have read and understand the issue reporting guidelines 

**Describe the bug**
when restarting the hotspot, sometimes the dhcp server will refuse to start, complaining about a bad line in:
/etc/dnsmasq.d/090_raspap.conf
that line being:
log-facility=/tmp/dnsmasq.log
when i comment this out and restart the hotspot i can connect again normally.
devices configured to use static ip connect fine.

**Your environment**
* rpi2
* Raspbian Buster Desktop
* using tl-wn722n

billz, 2020-07-15:
Unable to repro on RPi OS Buster Lite. Does /tmp exist on your filesystem?

1. Enable option DHCP Server > Log DHCP requests
2. Save DHCP settings
3. Restart hotspot 
4. Good AP + DHCP log output

```
Jul 15 09:45:17 dnsmasq[7148]: started, version 2.80 cachesize 150
Jul 15 09:45:17 dnsmasq[7148]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jul 15 09:45:17 dnsmasq-dhcp[7148]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
Jul 15 09:45:17 dnsmasq[7148]: reading /run/dnsmasq/resolv.conf
Jul 15 09:45:17 dnsmasq[7148]: using nameserver 192.168.1.254#53
Jul 15 09:45:17 dnsmasq[7148]: using nameserver fd0f:ee:b0::1#53
Jul 15 09:45:17 dnsmasq[7148]: read /etc/hosts - 5 addresses
Jul 15 09:45:18 dnsmasq[7148]: read /etc/raspap/adblock/hostnames.txt - 147846 addresses
```

Please note that Desktop is unsupported.

jginspace, 2020-07-16:
yes i have a /tmp, with an empty 'dhcpcd-pi' under it.

this problem hasn't reoccurred in the last 24 hours, even though i just changed the hotspot from g to n.

this is from the logs last time it happened:
`Jul 15 09:46:47 raspberrypi dhcpcd[24271]: wlan0: probing address 10.3.3.1/24
Jul 15 09:46:47 raspberrypi dhcpcd[24271]: eth0: soliciting an IPv6 router
Jul 15 09:46:47 raspberrypi dhcpcd[24271]: wlan0: soliciting an IPv6 router
Jul 15 09:46:48 raspberrypi dhcpcd[24271]: eth0: rebinding lease of 192.168.1.70
Jul 15 09:46:48 raspberrypi dhcpcd[24271]: eth0: leased 192.168.1.70 for 36000 seconds
Jul 15 09:46:48 raspberrypi dhcpcd[24271]: eth0: router 192.168.0.1 requires a host route
Jul 15 09:46:48 raspberrypi dhcpcd[24271]: eth0: adding route to 192.168.1.0/24
Jul 15 09:46:48 raspberrypi dhcpcd[24271]: eth0: adding host route to 192.168.0.1
Jul 15 09:46:48 raspberrypi dhcpcd[24271]: eth0: adding default route via 192.168.0.1
Jul 15 09:46:50 raspberrypi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jul 15 09:46:50 raspberrypi dnsmasq[24291]: dnsmasq: illegal repeated keyword at line 3 of /etc/dnsmasq.d/090_raspap.conf
Jul 15 09:46:50 raspberrypi dnsmasq[24291]: illegal repeated keyword at line 3 of /etc/dnsmasq.d/090_raspap.conf
Jul 15 09:46:50 raspberrypi dnsmasq[24291]: FAILED to start up`

as you can see it doesn't like me using a 192.168.1.0/24 network with a gateway of 192.168.0.1
i've also changed the wlan to 10.3.3.1 but this is recent.

this is from the last (successful) restart:
`Jul 16 09:02:55 raspap dhcpcd[11742]: wlan0: probing address 10.3.3.1/24
Jul 16 09:02:56 raspap dhcpcd[11742]: wlan0: soliciting an IPv6 router
Jul 16 09:02:56 raspap dhcpcd[11742]: eth0: rebinding lease of 192.168.1.70
Jul 16 09:02:56 raspap dhcpcd[11742]: eth0: leased 192.168.1.70 for 36000 seconds
Jul 16 09:02:56 raspap dhcpcd[11742]: eth0: router 192.168.0.1 requires a host route
Jul 16 09:02:56 raspap dhcpcd[11742]: eth0: adding route to 192.168.1.0/24
Jul 16 09:02:56 raspap dhcpcd[11742]: eth0: adding host route to 192.168.0.1
Jul 16 09:02:56 raspap dhcpcd[11742]: eth0: adding default route via 192.168.0.1
Jul 16 09:02:56 raspap dhcpcd[11742]: eth0: soliciting an IPv6 router
Jul 16 09:02:58 raspap systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jul 16 09:03:00 raspap dhcpcd[11742]: wlan0: using static address 10.3.3.1/24
Jul 16 09:03:00 raspap dhcpcd[11742]: wlan0: adding route to 10.3.3.0/24`

note 'soliciting an ipv6 router'.

billz, 2020-07-17:
Hmm, 'illegal repeated keyword' suggests that log-facility is defined elsewhere in /etc/dnsmasq.d/
If this happens again, check for it with `grep -r log-facility /etc/dnsmasq.d/`

jginspace, 2020-08-18:
yes, i even had a look in:
conf-file=/etc/raspap/adblock/domains.txt
addn-hosts=/etc/raspap/adblock/hostnames.txt

it was running without drama for a few weeks, then I had reason to add a dns server in the raspap gui under dhcp and it happened again. looking at the logs there i saw:
```
Aug 18 11:18:27 dnsmasq-dhcp[2627]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
Aug 18 11:18:27 dnsmasq-dhcp[2627]: DHCP, IP range 10.3.3.20 -- 10.3.3.99, lease time 12h
Aug 18 11:18:27 dnsmasq-dhcp[2627]: DHCP, IP range 10.3.3.20 -- 10.3.3.99, lease time 12h
Aug 18 11:18:27 dnsmasq-dhcp[2627]: DHCP, IP range 10.3.3.20 -- 10.3.3.99, lease time 12h
```
... as mentioned i changed my range to 10.3.3.x, so wondered where it was getting the (default) 141 from. the answer was /etc/dnsmasq.conf:
```
interface=wlan0
dhcp-range=10.3.141.50,10.3.141.255,255.255.255.0,12h
dhcp-option=6,192.168.1.53
log-facility=/tmp/dnsmasq.log
conf-dir=/etc/dnsmasq.d
```
obviously dnsmasq is reading this file first. i guess i should just comment out every line.

closed 2020-10-29 by stale[bot]