RaspAP/raspap-webgui #634: [Question] hotspot activation schedule

loloupolac, 2020-07-15:
Good morning all,

I apologize in advance, I don't know where I can ask questions about this great project that is raspap so I do it here. My question is quite simple: it is possible to manually deactivate the hotspot from the web menu but is it possible to define a hotspot activation schedule? For example, disable wifi on Monday and Tuesday between 02:00 and 08:00. This function is very common in consumer wifi systems but I don't know how raspap handles this.

Thanks for your help :)

billz, 2020-07-15:
There's no scheduler (yet) but it could be a nice feature to add. Until then, you can use cron to start/stop raspap's service control script at certain times. Run `sudo crontab -e` and add entries like so:

```
# Stop RaspAP services at 02:00 on Monday and Tuesday 
0 2 * * 1,2 sudo /etc/raspap/hostapd/servicestart.sh --action stop

# Start RaspAP services at 08:00 on Monday and Tuesday 
0 8 * * 1,2 sudo /etc/raspap/hostapd/servicestart.sh --seconds 3
```

loloupolac, 2020-07-15:
Thank you for this quick response! 
I will try this solution ;)

closed 2020-07-15 by loloupolac