RaspAP/raspap-webgui #635: Raspberry not using 2.4G but 5G with hotspot in g mode

sunberrysolar, 2020-07-16:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Choosing the wifi mode in hotspot show 2.4G for hotspot. But after restarting hotspot, it never use 2.4G but 5G.
Not really a bug, but the hotspot mode must not show 2.4G if it is not the case.

**Your environment**
* Pi 3 Model B
* Raspbian version (examples: Buster Lite, Buster Desktop)
* Followed the project prerequisites
* Checked the project FAQ
* RaspAP Quick Install
* Using default configuration without DHCP and OpenVPN
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset
* Other software or services running with RaspAP : Jeedom

**Steps to reproduce**

**Expected behavior**
Use 2.4G to connect, not 5G

**Actual behavior**
Use always 5G because it is connected to wificlient with 5G

Is there a way to force the Raspberry to use only 2.4G for wificlient and hotspot ?
I use TPLink HS100 and it does not have 5G, but only 2.4G

sunberrysolar, 2020-07-16:
[image]
[image]

Information about this issue (on RPi4) : https://raspberrypi.stackexchange.com/questions/101065/unable-to-create-2-4-ghz-wifi-access-point-on-the-raspberry-pi-4

billz, 2020-07-22:
You cannot operate in AP-STA mode simultaneously on two different frequencies.  
If you want a 2.4GHz AP, first configure your client on this frequency then enable the AP.

closed 2020-07-22 by billz