RaspAP/raspap-webgui #638: WebGUI not working after upgrade

swissueli, 2020-07-20:
First of all thanks a lot for the excellent work.

Did an upgrade from 2.4.1 to 2.5 with the new -upgrade function. 

WebGUI is no longer working in my case:

10.3.141.1 is currently unable to handle this request.
HTTP ERROR 500

Anything else is working fine - meaning my clients are working fine through the AP etc.

Thanks a lot for your help.

Ustrici, 2020-07-21:
Confirm the issue and noticed during the upgrade:
`Checking iptables rules
Rule already exits: -A POSTROUTING -j MASQUERADE
Rule already exits: -A POSTROUTING -s 192.168.**50**.0/24 ! -d 192.168.**50**.0/24 -j MASQUERADE
Enable RaspAP control service (Recommended)? [Y/n]: Y
RaspAP Install: Enabling RaspAP daemon
Disable with: sudo systemctl disable raspapd.service`
My network has another type of range => 192.168.**86**.0

billz, 2020-07-21:
@Ustrici please see [this](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#default-settings), thanks

commit 5c592cca (2020-07-21): Bugfix. Resolves #638
closed 2020-07-21 by billz