RaspAP/raspap-webgui #639: WiFi client AP mode greyed out with no indication of why

jasonblewis, 2020-07-23:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
I want to enable the wifi client on WLAN1. From what I can tell, I need to enable WiFi client AP mode however this is greyed out.
See screenshot
[image]

**Your environment**
* Raspberry Pi 4  
* Raspbian version Rasberry Pi OS
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y)
* Onboard wireless chipset or external adapter? Both
* Other software or services running with RaspAP?

billz, 2020-07-23:
> I want to enable the wifi client on WLAN1. From what I can tell, I need to enable WiFi client AP mode however this is greyed out.

The **WiFi client** UI is designed for this. AP-STA, aka wifi client AP mode is a different animal. More info [here](https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap).

jasonblewis, 2020-07-24:
Thanks, I've tried those instructions but they didn't work for me. I'll give it a go again though. thanks for the pointer

closed 2020-07-24 by jasonblewis

BinaryShrub, 2021-03-26:
I'm having the same problem, what was the fix?