RaspAP/raspap-webgui #641: VPN not working

julijus, 2020-07-24:
VPN not starting on raspberry pi zero w.
I follow all the installation instruction, and get: 

`Install OpenVPN and enable client configuration? [Y/n]: y
RaspAP Install: Installing OpenVPN and enabling client configuration
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  easy-rsa libccid liblzo2-2 libpkcs11-helper1 opensc opensc-pkcs11 pcscd
Suggested packages:
  pcmciautils openvpn-systemd-resolved
The following NEW packages will be installed:
  easy-rsa libccid liblzo2-2 libpkcs11-helper1 opensc opensc-pkcs11 openvpn pcscd
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,958 kB of archives.
After this operation, 5,437 kB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.
[ ✘ error ]  **Unable to install openvpn** 
Enabling openvpn-client service on boot
Failed to enable unit: Unit file openvpn-client@client.service does not exist.
[ ✘ error ]  **Unable to enable openvpn-client daemon** 
RaspAP Install: Creating OpenVPN control scripts
changed ownership of '/etc/raspap/openvpn/configauth.sh' from root:root to root:www-data
[ ✓ ok ] `

After installation RaspAP works except OpenVPN, after saving vpn settings:

Unable to save client auth credentialsx
Enabling auth-user-pass in OpenVPN client.confx
Updating auth-user-passx
Checking iptables rulesx
Rule already exits: -A POSTROUTING -o tun0 -j MASQUERADEx
Rule already exits: -A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPTx
Rule already exits: -A FORWARD -i wlan0 -o tun0 -j ACCEPTx
Unable to save OpenVPN client config

billz, 2020-07-24:
> `Do you want to continue? [Y/n] Abort. [ ✘ error ] **Unable to install openvpn**`

Looks like you aborted the installation process. Either run the installer again or use the manual steps.

closed 2020-07-24 by billz

julijus, 2020-07-26:
`Do you want to continue? [Y/n] Abort. [ ✘ error ] **Unable to install openvpn**`
At this step, there are no options to choose from [Y/n], automatically:
`Abort. [ ✘ error ] **Unable to install openvpn**`

feh123, 2020-07-26:
I have seen this automatic abort at install openvpn too. I have a seen it with both an +3B and RPI 4 running the Raspberry OS. It was not a constant error and sometimes openvpn installed without error.
Like you I found that the Raspap install would continue and complete. After completion I just installed openvpn myself and it worked fine. As it was working I never really tried to discover what caused the abort.

julijus, 2020-07-26:
> I have seen this automatic abort at install openvpn too. I have a seen it with both an +3B and RPI 4 running the Raspberry OS. It was not a constant error and sometimes openvpn installed without [...]

Thank, 
i  installed openvpn manual myself to , and it works!

ochen1, 2020-11-02:
Seems to be the same problem as this: #690, which my #701 should fix :)